maarten-pennings/CCS811 #1: constructor not working

holgerlembke, 2018-05-22:
Hi.

I see the idea you had with your constructor using default values. On the other hand, if you use CCS811() as a constructor call, this will call the default constructor, because it is still there and fits better than the constructor you created. 

So just a hint, that it might not work the intended way...

```
  public: // Main interface
    CCS811() {
		CCS811(-1,CCS811_SLAVADDR_0);
	}
    CCS811(int nwake=-1, int slaveaddr=CCS811_SLAVADDR_0);       
```

maarten-pennings, 2018-10-21:
Hi - sorry for the very late reaction.
I didn't known that github projects have active Issues list be default.

I tried to understand you comment, but I don't.
Maybe you can give me some more help.
I tried the attached test-cons.ino script, it gave the attached test-cons.txt output.
And that seems to do what I want.

Thanks for the feedback.

[test-cons.ino.txt](https://github.com/maarten-pennings/CCS811/files/2499414/test-cons.ino.txt)
[test-cons.txt](https://github.com/maarten-pennings/CCS811/files/2499415/test-cons.txt)

maarten-pennings, 2018-12-01:
Shall I close the issue?

holgerlembke, 2018-12-02:
yes please.

as far as I see, I'm wrong...

closed 2018-12-02 by holgerlembke