maarten-pennings/CCS811 #2: Temperature Compensation function for CCS881

KemleyBeharee, 2018-06-23:
Is it possible to have a function that takes in temperature and humidity parameters to improve readings

maarten-pennings, 2018-10-21:
Yes tha is possible.
The CCS811 has that feature; you could use the ENS210 for that (from same company as CCS811).
On my github you already find an ENS210 driver (but not yet incorporated in an CCS811 example).
I plan to write a second example using ENS210 - but hobby time is scarce :-)

KemleyBeharee, 2018-10-21:
Dear Maarten,

 

Thanks for reverting back

 

Regards

Dr. Kemley Beharee

Founder & CTO | Smart Health Ltd

www.smarthealth.mu

maarten-pennings, 2018-10-23:
Today I have added support for temperature and humidity compensation.
See example 'ccs811envdata' in the library,
I will close this `issue` hope that is ok for you.

closed 2018-10-23 by maarten-pennings