maarten-pennings/CCS811 #4: eco2=0 ppm,  etvoc=0 ppb

JohnnyPP, 2018-09-02:
Hi,

on ESP8266 I get always the same results of eco2=0 ppm and etvoc=0 ppb. The output looks like this:
CCS811: eco2=0 ppm,  etvoc=0 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=10 uA, raw10=407 ADC. The raw10 varies a little bit but stays in a range between 400 and 420. What could cause this behavior?

maarten-pennings, 2018-10-21:
Sorry for the late reply?
Is your sketch working now?
If not, could you also show the prints from the beginning of the sketch?
Since the D bit is never hi in STATUS, maybe the start command failed.

maarten-pennings, 2018-12-01:
Shall I close the issue?

bfaliszek, 2018-12-14:
I have a very similar problem. After uploading ccs811basic.ino I get ```CCS811: errstat = 10 = - vhxmrwf - Ad-ie``` or ``` CCS811: I2C error```. I am using ESP8266 (NodeMCU) and ccs811 on a purple PCB. Sometimes after several NodeMCU reboots, the code starts working and measurements appear.

maarten-pennings, 2018-12-14:
Can you show the entire output (from the beginning of the program)?

bfaliszek, 2018-12-15:
I changed the wires to other ones and now it works. I guess it was just a connection problem.

maarten-pennings, 2018-12-15:
Good you solved it 👍

closed 2019-01-07 by maarten-pennings