maarten-pennings/CCS811 #5: ESP8266 Thing

hsadeghilam, 2018-09-27:
I'm using ESP8266 Thing with CCS811 and I considering I/O # I made these changes to make the code run.

#define D1 14
#define D2 2
#define D3 5

But I'm getting this error. I have already tried connecting WAKE to GND and 5.

Starting CCS811 simple demo
ccs811: begin: reset failed
init: CCS811 begin FAILED
init: CCS811 versions FAILED
init: CCS811 start FAILED
CCS811: eco2=65535 ppm,  etvoc=65535 ppb,  errstat=3F9B=--VHXMRWF--AD-IE=ERROR|OLD,  raw6=63 uA, raw10=1023 ADC

saket424, 2018-09-28:
Try these. They worked for me

static const uint8_t D0 = 16;
static const uint8_t D1 = 5;
static const uint8_t D2 = 4;
static const uint8_t D3 = 0;
static const uint8_t D4 = 2;
static const uint8_t D5 = 14;
static const uint8_t D6 = 12;
static const uint8_t D7 = 13;
static const uint8_t D8 = 15;
static const uint8_t D9 = 3;
static const uint8_t D10 = 1;

Here is the serial output I saw when I dabbed a QTip with 50% Isopropyl Alcohol

CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=444 ADC
CCS811: eco2=404 ppm,  etvoc=0 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=443 ADC
CCS811: eco2=404 ppm,  etvoc=0 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=443 ADC
CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=446 ADC
CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=448 ADC
CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=450 ADC
CCS811: eco2=7992 ppm,  etvoc=1156 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=97 ADC
CCS811: eco2=7215 ppm,  etvoc=1038 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=106 ADC
CCS811: eco2=7992 ppm,  etvoc=1156 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=100 ADC
CCS811: eco2=665 ppm,  etvoc=40 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=349 ADC
CCS811: eco2=519 ppm,  etvoc=18 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=396 ADC
CCS811: eco2=462 ppm,  etvoc=9 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=8 uA, raw10=418 ADC
CCS811: eco2=418 ppm,  etvoc=2 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=8 uA, raw10=440 ADC
CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=98=--vhxmrwF--AD-ie=valid&new,  raw6=7 uA, raw10=394 ADC
CCS811: eco2=400 ppm,  etvoc=0 ppb,  errstat=90=--vhxmrwF--Ad-ie=ERROR|OLD,  raw6=7 uA, raw10=398 ADC

maarten-pennings, 2018-10-21:
Is your sketch working now, @hsadeghilam?

hsadeghilam, 2018-10-23:
I really did not get a chance to try this updated library. I actually ended up buying the CCS811 evaluation board from another manufacturer and used their library which worked fine for me!

maarten-pennings, 2018-11-04:
Ok - thanks for the update.
I close the issue.

closed 2018-11-04 by maarten-pennings