maarten-pennings/CCS811 #7: CCS811: I2C error

EnoraNedelec, 2018-11-30:
Hi, 

I am following the basic example from your library and running it on this [NodeMCU](https://fr.aliexpress.com/item/1PCS-Wireless-module-CH340-NodeMcu-V3-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/32665100123.html?spm=a2g0s.9042311.0.0.27426c37Su17Fw)  with this [CCS811](https://fr.aliexpress.com/item/CJMCU-811-CCS811-Carbon-Monoxide-CO-VOCs-Air-Quality-Numerical-Gas-Sensors/32782864505.html?spm=a2g0s.9042311.0.0.27426c37esGEx0). 

I have made the 2 edits in the `Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266`.  

But I am getting the error:  `CCS811: I2C error`.

This is my first project (except LED ones) with NodeMCU. So I am not sure what to do to solve this.  

Here is my circuit (I followed you color code for the wire):

[image]
[image]

I am powering the NoceMCU via USB

EnoraNedelec, 2018-12-02:
Thank you very much for your time Marteen ! 

Yes the first pictures were a mistake. I then used the correct pullups circuit and I still got the same error. 

I don't have access to the ouput from the  startup as I am not powering the NodeMcu from my computer so I don't have access to the Serial.print(). I use a remote debug over telnet with putty (I am still a beginner and I am afraid that powering my NodeMCU with my computer may break something in my computer).  

I couldn't make de debug print the things from the startup, instead I moved all these lines in the loop (cf my sketch below): 

Here is what I got (I didn't convert the version's values from hex, I simply printed them):

```
init: CCS811 begin  FAILED
init: hardware    version: 4294967295
init: bootloader  version: 4294967295
init: application version: 4294967295
init: CCS811 start measuring FAILED
```
And for the `cs811.read()`:
```
eco2=  65535  ppm
etvoc=  65535 ppb
raw 65535
errstat 16283
decode errstat=   1073675156
```
I tried several times and I always got the same values. 

If the CCS811  doesn't start, does it mean that it's broken ?

My [sketch ](https://github.com/EnoraNedelec/CCS811/blob/master/Sketch_with_outputing_in_loop.ino) with the outputing right at the beginning of the loop (look for the parts between ***) 

Thanks again for your help!

maarten-pennings, 2018-12-02:
If I were you, I would first concentrate on `begin`.

The very first line says `begin FAILED` and that means the begin, well, failed.
This means the CCS811 could not be put in "application mode", hence all next commands fail.
I think the problem is very early since even the versions come out wrong.
The number 4294967295 (decimal) is FFFF FFFF in hex, which means the SDA line is always hi, which means the CCS811 is not responding. The other numbers you show are all-1 in binary.

I would suggest you hook the NodeMCU+CCS811 to your PC.
You need to find out where the problem is, so keep the setup simple.
> I am afraid that powering my NodeMCU with my computer may break something in my computer

I understand that. I bought a 2 euro USB hub to isolate my NodeMCU from my PC.
By the way, the worst that can happen is a short circuit (between 5V and GND), and I know from experience that my Dell laptop can handle that :-) it shuts down the USB port, and it is back the next boot.

Also, start with just the CCS811 basic script. Again, keep the setup simple.

Please note that I made the `CCS811::begin()` in such a way that it should help you diagnose the problem. By default it checks many things (12 diagnostics), and prints out the problem. It would be helpful, if we have this print-out, it explains why `begin()` fails.

So, NodeMCU with CCS811 via USB hub to PC, run the standard CCS811 basic script, and capture the serial output.

maarten-pennings, 2018-12-02:
It could be as simple as the wrong I2C slave address.
To make the setup even simpler, you could ground nWAKE instead of wiring it to the ESP8266 (then the CCS811 is always woken up)

Gettyk, 2018-12-02:
To connect the nWAKE  to ground, should I modify this line?
`CCS811 ccs811(D3); // nWAKE on D3`

I tried to do this:

`CCS811 ccs811(GND);` 

But I am getting, this error in Arduino IDE: `'GND' was not declared in this scope`

maarten-pennings, 2018-12-02:
If you connect the nWAKE pin of the CCS811 to a GND pin directly, the CCS811 is always awake.
As a result, the driver no longer has to wakeup the CCS811.
You tell the driver that by passing -1 for the nwake pin.
The `ccs811.h` file tries to explain that (documentation in comments):
  
  `CCS811(int nwake=-1, int slaveaddr=CCS811_SLAVEADDR_0);                   // Pin number connected to nWAKE (nWAKE can also be bound to GND, then pass -1), slave address (5A or 5B)`

But this is only a small step to simplify setup. 
I'm much more interested in the prints of `begin()` :-)

Good luck!

Gettyk, 2018-12-02:
I connected the nWake to the GND and I am getting the same errors:

```
CCS811 begin  FAILED
hardware    version: 4294967295
bootloader  version: 4294967295
application version: 4294967295
CCS811 start measuring FAILED
```

Gettyk, 2018-12-02:
This is the output in the serial monitor for the basic example with nwake connected to ground:

```
Starting CCS811 basic demo
ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)
init: CCS811 begin FAILED
init: hardware    version: FFFFFFFF
init: bootloader  version: FFFFFFFF
init: application version: FFFFFFFF
init: CCS811 start FAILED
```

EnoraNedelec, 2018-12-02:
To connect the nWAKE to ground, should I modify this line?
`CCS811 ccs811(D3); // nWAKE on D3`

I tried to do this:

`CCS811 ccs811(GND);`

But I am getting, this error in Arduino IDE: 'GND' was not declared in this scope

I looked into ccs811.cpp and apparently I could do this (I don't get an error, but I am not sure, it's the right way to do it):
`CCS811 ccs811(-1);`

I connected the nWake to the GND and I am getting the same errors:

```
CCS811 begin  FAILED
hardware    version: 4294967295
bootloader  version: 4294967295
application version: 4294967295
CCS811 start measuring FAILED
```
Sorry I didn't see you first message, I am reading it now. I am doing the Basic example and connecting it right now to my computer. I will be back in 5 minutes.

This is the output in the serial monitor (I connected my nodemcu to my laptop usb) for the basic example with nwake connected to ground:

```
Starting CCS811 basic demo
ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)
init: CCS811 begin FAILED
init: hardware    version: FFFFFFFF
init: bootloader  version: FFFFFFFF
init: application version: FFFFFFFF
init: CCS811 start FAILED
```

maarten-pennings, 2018-12-02:
The crucial line here (with a typo: GNF=GND):
  `ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)`

This means there was no reaction on the used I2C slave address.
Note that `begin()` even tries the other I2C slave address, but also there was no reply.

This starts to sound like a hardware problem.

What scenarios do we have?
- Either the CCS811 is broken
- The wires between CCS811 and ESP8266 are broken, or wrong pins are used
- The ESP8266 I2C is broken

What other hardware do you have?
Do you have a scope, logic analyser, Saleae to check if there is signal on the I2C wires?
Do you have another I2C slave to check if your ESP8266 generates I2C signals on the wires?
Do you have another Arduino (another ESP8266 or even another micro) to check if the first ESP is broken?
[Note, some Arduino's are 5V and that is not good for the CCS811]

MagTun, 2018-12-02:
Thanks for taking the time to explain every steps of the debug process. 

I have change several times the cables and checked the continuity of the cables so I don't think the problem comes one there. 
I followed to pin out on the pictures, so apparently, the problem isn't from there either. 
I used a another NodeCMU (the same one as the one on the picture) and I get the same output: 

```
ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)
init: CCS811 begin FAILED
init: hardware    version: FFFFFFFF
init: bootloader  version: FFFFFFFF
init: application version: FFFFFFFF
init: CCS811 start FAILED
```

So it's likely the CCS811. 

I have other I2C devices like a LCD I2C (I have never use it, so I will have to find a tuto like this [one ](https://www.losant.com/blog/how-to-connect-lcd-esp8266-nodemcu) for that.)

I only have a basic multimeter.

maarten-pennings, 2018-12-02:
I'm back ...

This intrigued me:
> Also when using a I2C scanner I got this output:
> I2C device found at address 0x27 // (using SDA on D3 and SCL on D4) 

Does this mean you have an I2C scanner?
Is this a sketch running on ESP8266?
Using I2C on D1 and D2?
With an LCD hooked to this bus (D1/D2)?

That would be good - a firm basis.

Did you then do these steps?
- Disconnect LCD (SCL, SDA and VDD and GND)
- Connect CCS811 (SCL, SDA, VDD, GND and nWAKE to GND)
- Run the I2C scanner

This does _not_ report a device on address on 5A or 5B?

Then ... we might have a broken CCS811 board ...

maarten-pennings, 2018-12-03:
The  `I2C device found at address 0x5A  !` is a good sign.
But the 95% of the time getting `No I2C devices found. ` is not good.
Loose contact?
Do pull-ups help?

I bought the same board as you have:

[image]

And I hooked it up in the most simple way:

[image]

Zoom-in on the NodeMCU side:

[image]

Zoom in on the CCS811 side:

[image]

And then I ran the ccs811basic sketch.

    Starting CCS811 basic demo
    ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)
    init: CCS811 begin FAILED
    init: hardware    version: FFFFFFFF
    init: bootloader  version: FFFFFFFF
    init: application version: FFFFFFFF
    init: CCS811 start FAILED
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error

    ... reset ...

    Starting CCS811 basic demo
    ccs811: begin: ping failed (VDD/GNF connected? SDA/SCL connected?)
    init: CCS811 begin FAILED
    init: hardware    version: FFFFFFFF
    init: bootloader  version: FFFFFFFF
    init: application version: FFFFFFFF
    init: CCS811 start FAILED
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error
    CCS811: I2C error

    ... reset ...

    Starting CCS811 basic demo                                                      
    init: hardware    version: 12                                                   
    init: bootloader  version: 1000                                                 
    init: application version: 1100                                                 
    CCS811: errstat=0=--vhxmrwf--ad-ie                                              
    CCS811: errstat=0=--vhxmrwf--ad-ie                                              
    CCS811: waiting for (new) data                                                  
    CCS811: waiting for (new) data                                                  
    CCS811: waiting for (new) data                                                  
    CCS811: waiting for (new) data                                                  
    CCS811: waiting for (new) data                                                  

So, also instabilities.
I do have the two `// Maarten: added Clock stretching` lines added in si2c.c.
Hmm.

One difference is, of course, in the versions of the application: 1.1 on the Chinese board versus 2.0 on my ams board.

    init: hardware    version: 12                                                   
    init: bootloader  version: 1000                                                 
    init: application version: 1100                                                 

    init: hardware    version: 12
    init: bootloader  version: 1000
    init: application version: 2000

maarten-pennings, 2018-12-03:
I have run the script several times today. And actually only the very first time it failed. All other runs where successful.

But successful means: the `begin()` works, so there is i2c communication, but the measurements fail. I debugged this, somehow the STATUS in register DATA_RESULTS_ALG differs from the standalone status...

MagTun, 2018-12-03:
Hi Marteen, 

You are amazing! Your dedication is just incredible! 

I had try the I2C scanner with the pullups and it change nothing (still getting 95% "not device found), but  then I try again without the pull-up an that time,  I could get a stable I2C connection in a special angle so maybe soldering the pins might help. I will let you know once I do it. 

And again, thanks so much for your time!

maarten-pennings, 2018-12-04:
I made an update of the library; it now also supports "old" CCS811s, with firmware before 2.0.0.
Probably you have that in your "chinese board" (mine had).

MagTun, 2018-12-04:
You are just amazing!  

I didn't have the time to solder today, I will let you know when I will do it.

maarten-pennings, 2018-12-07:
I made an update to the library, you can now flash the CCS811 with 2.0.0 firmware (replacing the 1.1.0).
Use at your own risk...

MagTun, 2018-12-09:
I have seen your flash example, you are simply incredible! Thanks a lot for your time!

On my side, I am still unable to get a stable communication with the CCS811 (I am waiting to get a good solder iron).

closed 2019-01-18 by maarten-pennings