maarten-pennings/CCS811 #9: ccs811: begin: HW_ID read failed

Creepwood, 2019-01-12:
Something went wrong... but i don't know why. If I'm using the Ardafruit CCS811 lib, the sensor is working.

**My Code mods**
`CCS811 ccs811(-1, 0x5A); // nWAKE on D3`
`ccs811.set_i2cdelay(0); // Needed for ESP8266 because it doesn't handle I2C clock stretch correctly`

**My Hardware**
- CCS811 is like your picture "wire-ccs811only.jpg"
- WEMOS LOLIN32 (ESP32)

**Serial ouput:**
Starting CCS811 flasher
ccs811: begin: HW_ID read failed
init: CCS811 begin FAILED
init: hardware    version: FFFFFFFF
init: bootloader  version: FFFFFFFF
init: application version: FFFFFFFF
init: comment-out this code line if you want to flash
loop: ended ...

maarten-pennings, 2019-01-12:
Hi, I don't see any obvious mistakes. 

Did you connect nWAKE to GND (since you pass -1 in the constructor)?

Did you (can you) run the ccs811basic example? What does it say?

Creepwood, 2019-01-12:
Hi.
Yes, I connected nWAKE to GND.

**When I run the ccs811basic sketch I get following output:**
setup: Starting CCS811 basic demo
setup: library     version: 8
ccs811: begin: HW_ID read failed
setup: CCS811 begin FAILED
setup: hardware    version: FFFFFFFF
setup: bootloader  version: FFFFFFFF
setup: application version: FFFFFFFF
CCS811: I2C error

Now I'm going to try with a LOLIN D32... I will be back

Creepwood, 2019-01-12:
Same problem with LOLIN D32 :(

Creepwood, 2019-01-12:
Ok...now it works. And I can flash the new firmware :) Very happy!

I changed your function `CCS811::i2cread(int reg, int num, uint8_t * buf)` an copied in the code to I2Cwrite out of the Ardafruit CCS811 lib :D

**Now the function looks like:**

bool CCS811::i2cread(int reg, int num, uint8_t * buf)
{
	uint8_t value;
	uint8_t pos = 0;
	
	//on arduino we need to read in 32 byte chunks
	while(pos < num){
		
		uint8_t read_now = min((uint8_t)32, (uint8_t)(num - pos));
		Wire.beginTransmission((uint8_t)_slaveaddr);
		Wire.write((uint8_t)reg + pos);
		Wire.endTransmission();
		Wire.requestFrom((uint8_t)_slaveaddr, read_now);
		
		for(int i=0; i<read_now; i++){
			buf[pos] = Wire.read();
			pos++;
		}
	}
	return true;
}

<br>
<br>
<br>

sry for the bad markdown styling... I do not got it, how I can insert a code block

Creepwood, 2019-01-13:
For sure. I'm also interested :)

**I tested this code for you (removed while loop): And it works**
bool CCS811::i2cread(int reg, int num, uint8_t * buf)
{
	uint8_t value;
	uint8_t pos = 0;
	
	//on arduino we need to read in 32 byte chunks
		
	uint8_t read_now = min((uint8_t)32, (uint8_t)(num - pos));
	Wire.beginTransmission((uint8_t)_slaveaddr);
	Wire.write((uint8_t)reg + pos);
	Wire.endTransmission();
	Wire.requestFrom((uint8_t)_slaveaddr, read_now);
	
	for(int i=0; i<read_now; i++){
		buf[pos] = Wire.read();
		pos++;
	}
	return true;
}
**The Serial output:**
setup: Starting CCS811 basic demo
setup: library     version: 8
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: eco2=400 ppm  etvoc=0 ppb  raw6=3 uA  raw10=497 ADC  R=267054 ohm
CCS811: eco2=400 ppm  etvoc=0 ppb  raw6=3 uA  raw10=502 ADC  R=269741 ohm
CCS811: eco2=400 ppm  etvoc=0 ppb  raw6=3 uA  raw10=507 ADC  R=272428 ohm

<br><br><br>
**After that, I tested your function again, because maybe now with new frimware on ccs811 it will work: But it's the same error...**
// Reads 'count` bytes from register at address `regaddr`, and stores them in `buf`. Returns false on I2C problems.
bool CCS811::i2cread(int regaddr, int count, uint8_t * buf) {
  Wire.beginTransmission(_slaveaddr);              // START, SLAVEADDR
  Wire.write(regaddr);                             // Register address
  int wres= Wire.endTransmission(false);           // Repeated START
  delayMicroseconds(_i2cdelay_us);                 // Wait
  int rres=Wire.requestFrom(_slaveaddr,count);     // From CCS811, read bytes, STOP
  for( int i=0; i<count; i++ ) buf[i]=Wire.read();
  return (wres==0) && (rres==count);
}

**The Serial output:**
setup: Starting CCS811 basic demo
setup: library     version: 8
ccs811: begin: HW_ID read failed
setup: CCS811 begin FAILED
setup: hardware    version: FFFFFFFF
setup: bootloader  version: FFFFFFFF
setup: application version: FFFFFFFF
CCS811: I2C error

maarten-pennings, 2019-01-13:
Your working code looks like this at the moment , right?

```
bool CCS811::i2cread(int reg, int num, uint8_t * buf)
{
  uint8_t value; 
  uint8_t pos = 0;

  //on arduino we need to read in 32 byte chunks
  uint8_t read_now = min((uint8_t)32, (uint8_t)(num - pos));
  Wire.beginTransmission((uint8_t)_slaveaddr);
  Wire.write((uint8_t)reg + pos);
  Wire.endTransmission();
  Wire.requestFrom((uint8_t)_slaveaddr, read_now);

  for(int i=0; i<read_now; i++){
	buf[pos] = Wire.read();
	pos++;
  }
  return true;
}
```

Notice that `pos` is always `0` in the first part of the code.
And that in the second part `pos` equals `i`.
Furthermore, `num` is smaller than `32`, and `value` is not used.
Let's use all this to simplify your code further.
We get this:

```
bool CCS811::i2cread(int reg, int num, uint8_t * buf)
{
  Wire.beginTransmission((uint8_t)_slaveaddr);
  Wire.write((uint8_t)reg);
  Wire.endTransmission();
  Wire.requestFrom((uint8_t)_slaveaddr, (uint8_t)num);

  for(int i=0; i<read_now; i++){
	buf[i] = Wire.read();
  }
  return true;
}
```

Could you check if this simplification still works - I guess so (let's hope I made no typos).
It would be great if you could rename `reg` to `regaddr` and `num` to `count`, then the code is even more equal.

If you compare the above code to my driver code, there is not much of a difference left anymore:
 - your code has typecasts, I think they don't do much.
 - my code has a `delayMicroseconds()`, but I believe you have set that delay to 0us
 - My code reports errors, yours doesn't.
 - I thing the crucial difference is here `  Wire.endTransmission();` vs `int wres= Wire.endTransmission(false); // Repeated START`. 

The last bullet is tricky. You have no parameter in `Wire.endTransmission()`.
Then the parameter defaults to `true`. I have an explicit `false`.
This means I have a single start-write-read-stop I2C transaction (as is normal for a read from register), you have two I2C transactions start-write-stop then start-read-stop (which is tricky because the chip could loose the register address).

Could you try my code and replace the false in `Wire.endTransmission(false)` to a `true`.

You could also run as separate check the third bullet, and change my `return (wres==0) && (rres==count);` also to a `return true;`.

Happy debugging :-)

maarten-pennings, 2019-01-13:
Thanks for all the testing!
I already suspected the STOP condition.

We could change the codem however, from a transaction perspective my code (without the STOP) is better than the new code (with the STOP).

You are running this on an ESP32, right?
Could you try if the ESP8266 workaround works for ESP32 as well?

```
int wres= Wire.endTransmission(false); // Repeated START
delayMicroseconds(50); // <== extra wait here
```

Do you happen to have a logic analyser/scope?
I'm really interested to see what happens when this executes.

Creepwood, 2019-01-13:
No problem ;) Yes I am using a ESP32.
It seem, that the ESP8266 workaround is not working.

**1. try with `Wire.endTransmission(false)`: not working**
bool CCS811::i2cread(int regaddr, int count, uint8_t * buf) {
  Wire.beginTransmission(_slaveaddr);              // START, SLAVEADDR
  Wire.write(regaddr);                             // Register address
  int wres= Wire.endTransmission(false);           // Repeated START
  delayMicroseconds(50);                 // Wait
  int rres=Wire.requestFrom(_slaveaddr,count);     // From CCS811, read bytes, STOP
  for( int i=0; i<count; i++ ) buf[i]=Wire.read();
  return (wres==0) && (rres==count);
}
[image]

**2. try with `Wire.endTransmission(true)`: working**
bool CCS811::i2cread(int regaddr, int count, uint8_t * buf) {
  Wire.beginTransmission(_slaveaddr);              // START, SLAVEADDR
  Wire.write(regaddr);                             // Register address
  int wres= Wire.endTransmission(true);           // Repeated START
  delayMicroseconds(50);                 // Wait
  int rres=Wire.requestFrom(_slaveaddr,count);     // From CCS811, read bytes, STOP
  for( int i=0; i<count; i++ ) buf[i]=Wire.read();
  return (wres==0) && (rres==count);
}
[image]
<br>
I checked the whole scope buffer... but don't see any differents

maarten-pennings, 2019-01-18:
Sorry was a busy week - did not yet have time to look at this ...

maarten-pennings, 2019-01-24:
I think I solved your problem.
See the new section I added for [ESP32](https://github.com/maarten-pennings/CCS811/#esp32).
Basically, there is a bug in the ESP32 I2C lib (for repeated start conditions).
So, do not use ESP32 Arduino  core lib version 1.0.0, but use 1.0.1.

maarten-pennings, 2019-01-24:
Can I close the Issue?