maarten-pennings/CCS811 #10: somehow broken

coelner, 2019-01-24:
```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0d

Starting CCS811 flasher
setup: library     version: 10
setup: hardware    version: 12
setup: bootloader  version: FFFFFFFF
setup: application version: FFFFFFFF
setup: starting flash of 'CCS811_SW000246_1-00.bin' in 5 seconds

ccs811: ping ok
ccs811: reset ok
ccs811: status (reset1) 10 ok
ccs811: app-erase ok
ccs811: status (app-erase) 40 ok
ccs811: writing 5120 ................................................................ 4608
ccs811: writing 4608 ................................................................ 4096
ccs811: writing 4096 ................................................................ 3584
ccs811: writing 3584 ................................................................ 3072
ccs811: writing 3072 ................................................................ 2560
ccs811: writing 2560 ................................................................ 2048
ccs811: writing 2048 ................................................................ 1536
ccs811: writing 1536 ................................................................ 1024
ccs811: writing 1024 ................................................................ 512
ccs811: writing 512 ................................................................ 0
ccs811: app-verify ok
ccs811: status (app-verify) 30 ok
ccs811: reset2 ok
ccs811: status (reset2) 10 ok

loop: ended ...
loop: ended ...

load 0x4010f000, len 1384, room 16 
tail 8
c&⸮]⸮⸮⸮⸮D⸮b1.⸮⸮⸮e
Starting CCS811 flasher
setup: library     version: 10
ccs811: Not in boot mode, or no valid app: 20
setup: CCS811 begin FAILED
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
init: already has 2.0.0
loop: ended ...
loop: ended ...

setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
Scanning...
I2C device found at address 0x5A  !
done

ccs811: Not in boot mode, or no valid app: 20
setup: CCS811 begin FAILED
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: I2C error
CCS811: I2C error
CCS811: I2C error
```
I put a I2C Scan after using Wire.begin(), the device is there and up, I2C should be working. 
I'm currently on the master branch for esp8266 (commit a9e89500e16e1e9f07a3a4eec9370982e7b6b3cf) and using for this firmware flash a wemos D1 mini board.

Any suggestion?

maarten-pennings, 2019-01-25:
Actually I'm a bit puzzled.

In your first run, we see `setup: hardware    version: 12`. 
This means that not only `ccs811.begin()` was successful, but also `ccs811.hardware_version()`.
For me, this means correct wiring, working CCS811.
However, the next line `setup: bootloader  version: FFFFFFFF` is wrong; this looks like a failed read (no response from CC811, so SDA is always high, so ESP8266 thinks it's getting 1s).
But the next step `flash()` seems to work again.
Very strange.

The second run is reverse; now `begin()` fails but `hardware_version()` is ok.
Also very strange.

So, somehow you have unreliable communication.
Can you replace all wires?

I'm not sure if you read this "Unfortunately, the CCS811 uses clock stretching, and the I2C sw library in the ESP8266 (up to version 2.4.2) can not handle this"; can you check which version of the ESP8266 Arduino lib you have? (Tools > Board > Board manager, then eps8266. Did you do "My suggested do-it-yourself solution is described [here](https://github.com/maarten-pennings/I2C-tool/blob/master/I2Ctest8266/README.md#how-to-fix)".

Good luck, and let me know.

coelner, 2019-01-25:
The device is found with the address 0x54, the wiring is working. I changed the wires but no change at all.
For the problem with the clock stretching I use the current master branch because it contains several fixes especially for the ccs811.

i sniffed the traffic with a logic analyzer, you can open it with sigrok.
[ccs811_i2c.sr.zip](https://github.com/maarten-pennings/CCS811/files/2797065/ccs811_i2c.sr.zip)

I tested it with an arduino due board and it works after a reflash.

```
Starting CCS811 flasher
setup: library     version: 10
ccs811: Not in boot mode, or no valid app: 20
setup: CCS811 begin FAILED
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
setup: starting flash of 'CCS811_SW000246_1-00.bin' in 5 seconds

ccs811: ping ok
ccs811: reset ok
ccs811: status (reset1) 20 ERROR - ignoring
ccs811: app-erase ok
ccs811: status (app-erase) 40 ok
ccs811: writing 5120 ................................................................ 4608
ccs811: writing 4608 ................................................................ 4096
ccs811: writing 4096 ................................................................ 3584
ccs811: writing 3584 ................................................................ 3072
ccs811: writing 3072 ................................................................ 2560
ccs811: writing 2560 ................................................................ 2048
ccs811: writing 2048 ................................................................ 1536
ccs811: writing 1536 ................................................................ 1024
ccs811: writing 1024 ................................................................ 512
ccs811: writing 512 ................................................................ 0
ccs811: app-verify ok
ccs811: status (app-verify) 30 ok
ccs811: reset2 ok
ccs811: status (reset2) 10 ok

loop: ended ...
loop: ended ...
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
Scanning...
I2C device found at address 0x5A  !
done

setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: eco2=400 ppm  etvoc=0 ppb 
```
I close the topic. So far I would recommend not to use the master branch of the esp8266.

closed 2019-01-25 by coelner

maarten-pennings, 2019-01-25:
>  So far I would recommend not to use the master branch of the esp8266.
One last thing, which version of the matter branch should not be used?

coelner, 2019-01-25:
I was on commit a9e89500e16e1e9f07a3a4eec9370982e7b6b3cf