maarten-pennings/CCS811 #12: Very high eCO2 values

eried, 2019-02-05:
Hey! thanks a lot for the library, this is awesome. I saw the other thread with a similar question but my problem is slightly different:
[image]

It looks like an inverse burn :P haha for first days seemed to work on an acceptable range, but now the values are crazy high. Do I really need the humedity and temperature reference? 

I am using your example for ThingSpeak directly, no other modifications to the ESP8266.

abrahamMLG, 2019-03-31:
Hi !!!!
I have same problem, i conect with nano, burned 48 hours or more... when measures stop  growing a few hours in 7992 ppm was disconected. I´m not sure if module receives temperature and humidity levels.

` void setTemp(){
    
   h = dht.readHumidity()- ha; //Leemos la Humedad
   t = dht.readTemperature()- ta; //Leemos la temperatura en grados Centigrados
   taCCS= t-25;
   ccs.setEnvironmentalData(h, taCCS);

  } `

I´m not sure if this line is required:     float temp = ccs.calculateTemperature();  
Thanks 4 share!!!

eried, 2019-04-01:
I am still having the same issue but it is way less frequent now:

[image]

I am not sure I can trust those high readings after the initial period of burn

maarten-pennings, 2019-04-01:
Hi @eried, sorry, I missed your previous post. The sensor can be used without T and RH compensation. But the computed eCO2 is better _with_ setting the actual T and RH (the datasheet says, if I recall correctly, that the chip will use the default T=25C and RH=50%).

maarten-pennings, 2019-04-01:
Hi @abrahamMLG. I do not understand the code you listed. The driver function is called
  CCS811::set_envdata(t,h)
not
  ccs.setEnvironmentalData(h, taCCS)

abrahamMLG, 2019-04-01:
Hi!, u right, mistake the sketch, sorry... I also try adafruit library and code corresponding at this sketch. But I check yours, upload and try again with same results... I don't know why, but when ccs is running a few hours increase the eco2 measure... My English is not so good and I can't translate to understand everything.. I will try to check baseline but I can understand the process... I don't wanna brick the ccs...

Thanks soooo much 4 your contribution with CCS811. It's soooo helpful.

Regards
Abraham

abrahamMLG, 2019-04-02:
Hi Erwin! I can't understand why, I try adafruit library and have same results, I have 4 ccs... And all of them make the same... Maybe I don't send environment data properly. I will check the code again and tell u something if I can fix the problem.

Regards
Abraham

[image]<https://user-images.githubusercontent.com/1091420/55313366-df865c00-5467-11e9-8bc6-99a08badd9d3.png>

eried, 2019-04-02:
> Hi @eried, sorry, I missed your previous post. The sensor can be used without T and RH compensation. But the computed eCO2 is better _with_ setting the actual T and RH (the datasheet says, if I recall correctly, that the chip will use the default T=25C and RH=50%).

Oh... I am going to try with those defaults. Thanks

abrahamMLG, 2019-04-02:
Hi again,  i think have solution to my problem, testing now... 
I use DHT sensor,  and use softwareserial to control data wit BT app.
Hope found the problem, i was using reading code in two places in the sketck, maybe this was the problem, now pass eco2 value to another variable and work with that. 

`/*
  ccs811envdata.ino - Reading temperature and humidity data from the ENS210, passing that to the CCS811, to get better CO2 readings from CCS811
  Created by Maarten Pennings 2018 oct 23
*/

#include <SoftwareSerial.h>
#include <Wire.h>    // I2C library
#include "DHT.h"   // Libreria sensor humedad y temperatura
#include "ccs811.h"  // CCS811 library
#define DHTPIN 2     // Pin donde está conectado el sensor DHT
#define DHTTYPE DHT11   // Descomentar si se usa el DHT 11

  SoftwareSerial BT1(7,8);
  CCS811 ccs811(13); // nWAKE on 13
  DHT dht(DHTPIN, DHTTYPE);

  int h, t;
  int ha=1;    // adjustment of the reading deviation in DHT comparing with the real measures
  int ta=0.6;  // adjustment of the reading deviation in DHT comparing with the real measures
  bool ok;
  int co2;
  char control;
  
void setup() {

  BT1.begin(9600);
  // Enable serial
  Serial.begin(115200);
  Serial.println("");
  Serial.println("setup: Starting CCS811 envdata demo");
  Serial.print("setup: library     version: "); Serial.println(CCS811_VERSION);

  // Enable I2C, e.g. for ESP8266 NodeMCU boards: VDD to 3V3, GND to GND, SDA to D2, SCL to D1, nWAKE to D3 (or GND)
  Serial.print("setup: I2C ");
  Wire.begin(); 
  Serial.println("ok");
  
  // Enable ENS210
  // Serial.print("setup: ENS210 ");
  // ok= ens210.begin();
  // if( ok ) Serial.println("ok"); else Serial.println("FAILED");

   // Enable dht
   dht.begin();
  // Enable CCS811
  Serial.print("setup: CCS811 ");
  ccs811.set_i2cdelay(50); // Needed for ESP8266 because it doesn't handle I2C clock stretch correctly
  ok= ccs811.begin();
  if( ok ) Serial.println("ok"); else Serial.println("FAILED");

  // Start CCS811 (measure every 1 second)
  Serial.print("setup: CCS811 start ");
  ok= ccs811.start(CCS811_MODE_1SEC);
  if( ok ) Serial.println("ok"); else Serial.println("FAILED");
  delay(1000);
}

void loop() {
  // Read ENS210
  //int t_data, t_status, h_data, h_status;
  //ens210.measure(&t_data, &t_status, &h_data, &h_status );

  /* / Process ENS210
  Serial.print( "ENS210: " );
  if( t_status==ENS210_STATUS_OK && h_status==ENS210_STATUS_OK ) {
    Serial.print("T="); Serial.print( ens210.toCelsius(t_data,1000) );     Serial.print(" mC  ");
    Serial.print("H="); Serial.print( ens210.toPercentageH(h_data,1000) ); Serial.print(" m%RH  ");
    // Pass environmental data from ENS210 to CCS811
    ccs811.set_envdata210(t_data, h_data);
  } else {
    Serial.print( "T=" ); Serial.print( ens210.status_str(t_status) ); Serial.print("  ");
    Serial.print( "H=" ); Serial.print( ens210.status_str(h_status) ); Serial.print("  ");
  }
  Serial.print("  ");
*/

  // Read CCS811
  uint16_t eco2, etvoc, errstat, raw;
  ccs811.read(&eco2,&etvoc,&errstat,&raw);
   
  // Read DHT
   h = dht.readHumidity()- ha; //Leemos la Humedad
   t = dht.readTemperature()- ta; //Leemos la temperatura en grados Celsius
   ccs811.set_envdata210(t, h);  // Send data to ccs811
   
  // Process CCS811
  
    Serial.print("CCS811: ");
    if( errstat==CCS811_ERRSTAT_OK ) {
    Serial.print("eco2=");  Serial.print(eco2);  Serial.print(" ppm  ");
    Serial.print("etvoc="); Serial.print(etvoc); Serial.print(" ppb  ");  
  } else if( errstat==CCS811_ERRSTAT_OK_NODATA ) {
    Serial.print("waiting for (new) data");
  } else if( errstat & CCS811_ERRSTAT_I2CFAIL ) { 
    Serial.print("I2C error");
  } else {
    Serial.print( "error: " );
    Serial.print( ccs811.errstat_str(errstat) ); 
  }
  Serial.println();
  co2=eco2;
  envdatos()

if (BT1.available()){
    
    envdatos();
    }
  // Wait
  delay(2000); 
}

void envdatos(){  
  if (BT1.available()){
   control=BT1.read();
   if (control=='L'){                    //envia datos
   BT1.print("temperatura: ");BT1.println(t);
   BT1.print("humedad: ");BT1.println(h);
   BT1.print("Co2: ");BT1.println(co2);
   BT1.flush();
   co2=0;
   delay(2000);
     }
  }  
  } `

Regards

maarten-pennings, 2019-06-02:
Hi @eried, I closed this issue, hope it is OK with you.

closed 2019-06-02 by maarten-pennings