maarten-pennings/CCS811 #15: Version 1100 wrong ERROR_ID read?

iKlask, 2019-04-18:
I had an issue with my cjmcu-811 on firmware 1100. It kept throwing the following at startup:
```
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 1100
CCS811: waiting for (new) data
CCS811: errstat=91=--vhxmrwF--Ad-iE
CCS811: errstat=99=--vhxmrwF--AD-iE
CCS811: errstat=99=--vhxmrwF--AD-iE
.
.
.
```
This seemed odd because it meant it was in application mode (good), application firmware loaded (good), data ready (good), and some error in the ERROR_ID... but none of the other flags were being thrown even though it thinks there's an error in ERROR_ID?

So I took out the debugger and found this code block:
```
    if( _appversion<0x2000 ) {
      ok= i2cread(CCS811_STATUS,1,&stat); // CCS811 with pre 2.0.0 firmware has wrong STATUS in CCS811_ALG_RESULT_DATA
      if( ok && stat==CCS811_ERRSTAT_OK ) ok= i2cread(CCS811_ALG_RESULT_DATA,8,buf); else buf[5]=0;
      buf[4]= stat; // Update STATUS field with correct STATUS
    } else {
      ok = i2cread(CCS811_ALG_RESULT_DATA,8,buf);
    }
```
The important part is the third line. If i2cread() passed and stat==CCS811_ERRSTAT_OK, then read the whole CCS811_ALG_RESULT_DATA into buff. Otherwise... just zero out the ERROR_ID part of buff at buf[5]? Why?

My issue ended up being a poorly soldered zero-ohm resistor between pins 4 and 5 on the CCS811 package. These are the PWM and Sense pins which need a connection for sensing the Heater current. Since these were not connected the board was throwing a `CCS811_ERRSTAT_HEATER_SUPPLY` error in ERROR_ID but I was never getting these flags printed for me. Correct me if I'm wrong, but it seems as if your code doesn't read the ERROR_ID byte unless there are no errors reported by status? If there's an error in status, it doesn't read anything into buff.

A second note, this whole workaround for v1100 is supposedly because the status register from `i2cread(CCS811_ALG_RESULT_DATA,8,buf);` reads incorrectly in firmware v1100. However, my status register seemed to read fine on firmware v1100 from `CCS811_ALG_RESULT_DATA`.

maarten-pennings, 2019-04-21:
Hi @iKlask 
You might be right, the buf[5]=0 is fishy.
Do you have a proposed fix?

dodo041, 2022-01-25:
Hi @maarten-pennings, I'm working with an adaptation of your code for the CCS811 sensor, written in plain C. Checking the logics, I came across the same issue which was presented by the author. I'm thinking about a fix for the issue, since the else case in line 
`if( ok && stat==CCS811_ERRSTAT_OK ) ok= i2cread(CCS811_ALG_RESULT_DATA,8,buf); **else buf[5]=0;**`
definitely seems wrong, since there's no point in overwriting the byte buf[5] containing the error information, when the status byte suggests the existence of an error.

My idea was to read the error id from the error register, when the error bit in the status register is set. So (as before) you read the status register only, check for existing errors, and if the error bit in the status register is set, read the error register e.g. with
`ok= i2cread(CCS811_ERROR_ID,8,buf);` 
(with CCS811_ERROR_ID being the address 0xE0 in the application register), and then just proccessing the error id as needed.
 Afterwards replace the  status  byte from  ALG data with the previously read "correct" status, and if needed the errror id byte with  the read error id, otherwise (if  there's no error and the  error id  byte is also errnoeous , set it to 0).

Before going along with this simple fix, I'd like you to supply me with some information (if you still can remember, since the issue is opened for almost 3 years now):
- What was the issue you encountered with the returned status byte, when using firmware version < 2.0.0? @iKlask pointed out, that he did not have problems with versions lower than that. Maybe I'm leaving  something obvious and important out of my considerations, but for now this seems like the only  plausible way of an workaround.

Groetjes in het mooie Nerderland!