maarten-pennings/CCS811 #18: Error Code

delsol25, 2019-05-20:
Hello Maarten,
First, many thanks for your code, this is really appreciated.
I'm not able to get any value out of my CCS811.
First I upgraded the application to 2.0 using your code. It work without any problem.
Here is what I'm getting: 

setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: waiting for (new) data
CCS811: errstat=2091=--VhxmrwF--Ad-iE
CCS811: errstat=2091=--VhxmrwF--Ad-iE
CCS811: errstat=2891=--VhXmrwF--Ad-iE
CCS811: errstat=2899=--VhXmrwF--AD-iE
CCS811: errstat=2899=--VhXmrwF--AD-iE
CCS811: errstat=2899=--VhXmrwF--AD-iE
CCS811: errstat=2899=--VhXmrwF--AD-iE

I updated to the latest ESP8266 Library.
I'm using a China D1 Mini
Supply is maybe a little low 3.27 since i'm powering with the USB.
Wake pin is tie to GND.

Waiting for your feedback
David

delsol25, 2019-05-20:
Hello again,
Really sorry to distrub, I think I just got a bad sensor. I tested with another one and everything is fine.
I also updated the firmware . 2 on 2.
Thanks

maarten-pennings, 2019-05-20:
As you can see in the .h file
ERRSTAT is a merge of two hardware registers: ERROR_ID (bits 15-8) and STATUS (bits 7-0)

So, we have to split the hex value 2899 and the string (--VhXmrwF--AD-iE).
ERROR has hex value 28 which corresponds to --VhXmrw  and
STATUS has hex value 99 which corresponds to F--AD-iE
Capitals mean bit-is-set, lowercase means bit-is-clear, dash means bit has no meaning.
See function CCS811::errstat_str for details

What does this mean?

Most part of STATUS is ok (see datasheet page 18 on ams.com/ccs811): The F tells that the firmware is in application mode (not bootmode) , the A means that the application image is valid (flashed correctly), D means there is new measurement data. But you also have an E - for Error. This should not happen.

When you have an E/error, the ERROR register gives details. You have V and X (datasheet page 24); X meaning "The sensor resistance measurement has reached or exceeded the maximum range". I'm not sure what does means, maybe your sensor is indeed broken.

maarten-pennings, 2019-05-20:
If you are happy, you can close the issue.

closed 2019-05-20 by delsol25