maarten-pennings/CCS811 #21: Reflashed Adafruit CCS811 and lost access to thermistor

bill-orange, 2019-05-31:
The reflash of an Adafruit CCS811 to 2.0.0.0 went well.  No worries there.  The little chip now reports being at version 2.0.0 from version 1.  However to my surprise I can no longer see the onboard thermistor.  I do not have a copy of the old binary so I can not go back (not that I want to).  Should I use a different binary for on-board thermistor support?

bill-orange, 2019-05-31:
OOPS.  This is from the AMS web site (bold face my edit):

CCS811 Firmware - Revision History

Version    Changes

2-0-1      Firmware build including all 2-0-0 features accept management of the burn in period          

2-0-0      MOX sensor need a burn-in period of several days of operation from first power on before 
                 eCO2 and TVOC readings stabilize. The burn-in period is now managed so that stable 
                 readings are available after only 60 minutes of operation after first power on
           Extend eCO2 maximum output value to 64000 ppm
           Extend TVOC maximum output value to 64000 ppm
           **Removed NTC functionality.** Pin 8 not measured and left undriven
           Added "Internal_State" variable to the command register map
           Improved the algorithm which computes eTVOC and eCO2 

1-1-0      Initial Version

maarten-pennings, 2019-06-02:
Hi @bill-orange, 
I checked the zipfile https://ams.com/documents/20143/36005/CCS811_SW000246_2-00.zip 
(click on 'CCS811 Application firmware version 2.0.0' at https://ams.com/ccs811#tab/tools).

The `Revision_History.txt` inside says
>    2-0-1      Firmware build including all 2-0-0 features accept management of the burn in period          

The  `readme.txt` inside says
> The version of firmware to use depends on the usage status of the device. 
> 	New fresh sensors use firmware 2-0-0 
> 	Sensors run for a number of days use firmware 2-0-1

I interpret this as follows.
- the two firmware images (2-0-0 and 2-0-1) are identical
- the one exception: 2-0-1 skips the burn-in correction period

This means that if you use 2-0-0 on a _used_ sensor, the first ~5 days its reported values are suboptimal because burn-in correction is applied. But after the 5 days, both firmware are the same again.

So, to keep things simple, I propose we stick to the 2-0-0.
It is tested, and 2-0-1 would be incorrect for fresh sensors...

If you really want to switch to 2-0-1, download and unzip the zip file mentioned above, run hex.py on `CCS811_FW_App_v2-0-1.bin` to generate the h-file, and use the generated h file instead of CCS811_SW000246_1-00.h.

bill-orange, 2019-06-02:
Makes sense to me. I was going to run hex.py but I ran into problems.  I am using Eclipse and get this error.

  File "C:\Users\William\workspace\hex conversion\hex.py", line 32
    if( n%16==0 ): print( "  ", end="" )
                                   ^
SyntaxError: invalid syntax

If I delete that line (and similar) for testings, I find that the byte.hex() conversion to hex fails with the .hex function missing.

What environment python environment are you using?  It sure must be different than mine!

Suggestions?

maarten-pennings, 2019-06-02:
Interesting ... 
The parenthesis used in the 'if' are not needed (unlike in C or javascript).
But parenthesis (usually) thet don't harm: 2+3 is the same as (2+3).

I just tried this

> C:\Users\maarten>python
> Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:01:18) [MSC v.1900 32 bit (Intel)] on win32
> Type "help", "copyright", "credits" or "license" for more information.
> >>> if( 2==2 ) : print("test")
> ...
> test

So, I'm guessing here, that you still run python 2.x, and I'm on 3.x.

bill-orange, 2019-06-02:
You are correct I am on 2.0..  There are many other errors to get past so, I will try running it on one of my PIs.  they have both 2.x and 3.x

bill-orange, 2019-06-02:
mission accomplished .
[CCS811_FW_App_v2-0-1.zip](https://github.com/maarten-pennings/CCS811/files/3244998/CCS811_FW_App_v2-0-1.zip)

closed 2019-06-02 by bill-orange