maarten-pennings/CCS811 #23: CJMCU-811 successful flash

msbjx5, 2019-06-26:
Hi Maarten how are you man ?
I just successful flashed my CJMCU-811 using an WEMOS D1

Thank you so much man

msbjx5, 2019-06-28:
Hi Maarten how are you ?
I flashed the CJMCU-911 it was fine, but now the sensor only works when i use your library.
When i try to upload your sketch ccs811basic to my heltec v2 i get the following error:

ccs811basic:12:15: error: 'D3' was not declared in this scope
 CCS811 ccs811(D3); // nWAKE on D3

Could you give me some help ?

Thanks

maarten-pennings, 2019-06-28:
Hi, D3 is the name of a constant. This name is apparently not defined for you. It should be defined, because it identifies a pin of the esp8266. My guess is that you are using the Arduino environment, but that you selected the wrong board. Go to Tools, then Board, then select your board (wemos d1?). Then it should compile.

msbjx5, 2019-06-29:
Hi man, thanks for your reply.
You're right I am using the Arduino environment.
I used my Wemos D1 to flash the sensor, but now I am trying to use it on my other board that is a Heltec WiFi LoRa 32 (V2), I have selected the right board under Arduino IDE but the error persists ( I hooked up the wires following as the wire-ccs811only.jpg ).
Do the device address change when we flash it ? I found out that i can use Sparkfun library again but only when i change the address from 0x5B to 0x5A. The Adafruit library used to work but it's not working anymore even when i change the device address in it. It starts showing: ERROR.
Thanks man

maarten-pennings, 2019-06-29:
So, you switched hardware to WiFi LoRa 32 (V2).
And you say you configured the arduino (Tools>Boards) to use this board.
Perfect.

However, the WiFi LoRa 32 (V2) is an ESP32 based board.
And the ESP does not have a D3 pin.
So, where the ESP8266 board has a definition for D3, the ESP32 board does not have it.
So you get a compiler error (`ccs811basic:12:15: error: "D3" was not declared in this scope`).

There are two solutions.

- The lazy one, use 
  `CCS811 ccs811(-1); // always wake`
  The driver will never put the CCS811 to sleep, so it does not need a wire to the wake-up pin.
- The energy saving one. 
  Pick a GPIO pin on the ESP to use for wakeup, say GPIO23. 
  Put a wire from the ESP G23 to nWAKE of CCS811, and replace the C source line with 
  `CCS811 ccs811(23); // nWAKE on GPIO23`

maarten-pennings, 2019-06-29:
By the way, I'm not aware that the firmware update changes the default I2C address.
Did you maybe treat the CCS811 ADDR pin differently on the two boards?

msbjx5, 2019-06-30:
Hi man, both solutions worked like a charm, I was able to upload the sketch to the heltec board.
The serial monitor starts with:
```
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: waiting for (new) data
CCS811: eco2=400 ppm  etvoc=0 ppb  
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
```
I only can get readings when the CO2 is above 400ppm under that I get errstat

Thanks for your time helping me

maarten-pennings, 2019-06-30:
Hmm. Does not look good. 

The E at the end tells there is an error. The first 8 chars then detail which error. In your case that is X which means CCS811_ERRSTAT_MAX_RESISTANCE.

Looking at the datasheet it says "The sensor resistance measurement has reached or exceeded the maximum 
range". I guess this means your sensor gives a value that is our of spec, which might indicate a broken sensor...

msbjx5, 2019-07-01:
OK, I'll buy another one.
Thanks for you help man !

closed 2019-07-05 by msbjx5

SFeli, 2019-12-29:
Hi M.
flashed my CCS-811 on my CJMCU-8128 with Arduino ESP32 without problems.
thx
Stefan

maarten-pennings, 2020-01-05:
Thaks Stefan, I have added you to the success table!

glychee, 2020-02-12:
I came here because i have the exact same issue as msbjx, using other libraries (I tried at least 4 others) my sensor gives readouts of 3 times 0, then stuck on 400 continuously. It used to work the first time i attached it, now not at all. 

I did breathe out directly onto the sensor to check it's upper limits when it did work and exceeded co2 of 7800 something, maybe this damaged my sensor?

Anyway, here's my serial output 
```
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
CCS811: waiting for (new) data
CCS811: errstat=1091=--vHxmrwF--Ad-iE
CCS811: errstat=1091=--vHxmrwF--Ad-iE
CCS811: errstat=1891=--vHXmrwF--Ad-iE
CCS811: errstat=1899=--vHXmrwF--AD-iE
CCS811: errstat=1899=--vHXmrwF--AD-iE
```
I did my reflashing using an M5StickC (orange esp32 based dev board) and it seems to have worked fine, as the application version is now 2000, maybe interesting for your success list! However this doesn't solve a broken sensor, if this is the case of course.

maarten-pennings, 2020-02-13:
Hi, the uppercase H indicates a heater error. So indeed maybe your sensor is broken.

I do not believe that breathing on it can brake it. Maybe temporary strange freakout readouts, but not permanent damage.

Also note that internally, the chip is a physical transducer (heater plus mox layer) plus a microcontroller. So upgrading firmware will work als long as the micro is ok.  In your case, the H indicates the heater in the transducer might be broken.

By the way, i did not understand your remark about the i2c address.

Good luck.

glychee, 2020-02-13:
Msbjx5's comment on 29th of June mentions his i2c address being on 0x5A. I wanted to confirm that reflashing does not change the address since I own the exact same chip as he does, which ships with an i2c address of 0x5A instead of 0x5B. Thus his statement regarding that to use the adafruit library he had to change the address in code is not something that's caused by an update.

Thanks for the reply! Now that you mention, another library did also tell me there was a heater error. When I receive my next chip I'll immediately reflash using this library and try to be more careful with it.

maarten-pennings, 2020-02-14:
The i2c address is determined by the internal micro. So indeed, in theory a firmware update could change the i2c address. However, I strongly believe this will never happen. The datasheet specifies that the chip can have two addresses, and you select which one with the ADDR pin. It would be very unwise to change that spec point.

Which of the two i2c addresses is used depends on the ADDR pin. That pin is tied to Vcc or GND by the board you purchase. 

So in practice the board determines the i2c address.

bradshawlupton, 2021-07-16:
Hi, Bradshaw here in Buzzards Bay, Massachusetts.

is there a listing of the firmware 2000 so that I may read the error numbers, letters, etc

I tried reset, reloading the source (on esp8266 with and without pull-up.

It ran great until I gave the pair of CCS811 and demos d1 esp8266 a day of idle.

----------------  Great work, just another little speed bump between here and perfection! tnx

18:38:45.797 -> setup: pi@piRshared.org STEM Fresh And Dirty Air Detective Agency
18:38:45.905 -> setup: Starting CCS811 basic demo
18:38:45.905 -> setup: ccs811 lib  version: 10
18:38:46.001 -> in begin
18:38:46.001 -> wake_up
18:38:47.853 -> setup: hardware    version: 12
18:38:47.886 -> setup: bootloader  version: 1000
18:38:47.985 -> setup: application version: 2000
18:38:51.834 -> CCS811: waiting for (new) data
18:39:01.856 -> CCS811: waiting for (new) data
18:39:11.842 -> CCS811: waiting for (new) data
_**18:39:21.857 -> CCS811: errstat=891=--vhXmrwF--Ad-iE
18:39:31.848 -> CCS811: errstat=899=--vhXmrwF--AD-iE**_

       and I got an 891 AND 899 differing in the upper and lower case d.

       an error or firmware list would be an amazing addition.   thank you

maarten-pennings, 2021-07-17:
Hi, as it happens, I have appended a section "Error flags overview" to the home page https://github.com/maarten-pennings/CCS811.

The bad news is that you have E flagged, which means error. Then the upper flags tell you which. You have X: "CCS811_ERRSTAT_MAX_RESISTANCE The sensor resistance measurement has reached or exceeded the maximum range". That is a hardware fault...

bradshawlupton, 2021-07-17:
out of interest, being a tinkerer, I have the sensors on breadboards in a plastic shoebox, I placed a 4 oz glass and half filled it with seltzer water.  it appears both sensors work if there is >400 (greater than something??) CO2 level in there.

so this is lesss broken than it looks.  if the firmware is available. I would be happy to look and maybe make a nice change, I have multiple decades of fun firmware working on my many bugs.  It would be neat to see the implementation.  I would happily NDA with anyone.  I want to base a product for STEM students 8th graders in the US.  So that's what's "in it for me"

THanks Bradshaw

I have been working in int and gas chromatography since college. SO I am very interested in doing a lab grade calibrations that a student could do with baking soda, measured, and vinegar measured, in a standard hardware store $2 5 gallon (20L) closed bucket.

So it is NOT failed, it is just sick. maybe a virus of some kind in the firmware (ha ha)

[image]

bradshawlupton, 2021-07-17:
iw there a way to "return to factory defaults' maybe reset the baseline?

I have another I will read the base registers and see it I can trick the ones I have back into normal operation.

bradshawlupton, 2021-07-18:
I let it run about another day of errors then did the seltzer water test, and it pops right into operation. this "feels" like a calibration problem to me.

[image]

I have been building calibration routines for VOC's and specific identifiable organics and inorganic for quite a while.  (the seltzer water is cranberry lime polar seltzer water, with a nice flavor/smell which accounts for the tvoc.   I have a local chrootagraphy lab where I worked recently, so I can calibrate the TVOC with isobutylene and make a correction curve as well.

this would be neat to add to the SW, or the firmware.

Bradshaw

maarten-pennings, 2021-07-18:
If you subject the sensor to a single gas, it is "easy" to calibrate. In real life, the "air" is always a mixture, unknow of what types of gasses, unknown in what ratios, and each of them with a different excitation curve (which might depend on eg temperature). On top of them, some gasses, like CO2 (!), do not even excite the sensor.

maarten-pennings, 2021-07-18:
> is there a way to "return to factory defaults' maybe reset the baseline?
There is no "go to factory defaults", since there is no persistent storage.
The baseline is in RAM, so each time you power cycle, the chip is in "factory defaults".
That is, by the way, why there is a get/set baseline (relying on the persistent storage of the host).

The above is a bit of a lie: there is persistent storage in the CCS811.
It records the operation hours during burn-in. To the best of my knowledge this can not be "factory reset".
Which makes sense, because once the MoX layer is "baked" it is "baked".

bradshawlupton, 2021-07-18:
thanks for the persistence of only the burnin time (hence the 2001 firmware.

do these hardware errors strongly suggest that I RMA return merchandise authorization swap them out with Adafruit?  

They only have a few days use on them.  I wanted them to run for a couple of years.

maarten-pennings, 2021-07-18:
Interesting question. Maybe you can ask adafruit...