maarten-pennings/CCS811 #24: Sleep mode

apaczola, 2019-06-29:
Hi! Thanks a lot for the library, it's working fine with ESP32, but i have a problem when i try to use it with sleep mode.

My code is:

`#include <Wire.h>    // I2C library
#include "ccs811.h"  // CCS811 library

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  60             /* Time ESP32 will go to sleep (in seconds) 1 minute */

CCS811 ccs811(23);

void setup() {
  Serial.begin(115200);
  Wire.begin(); 
  
  // Enable CCS811
  ccs811.set_i2cdelay(50); // Needed for ESP8266 because it doesn't handle I2C clock stretch correctly
  bool ok= ccs811.begin();
  if( !ok ) Serial.println("setup: CCS811 begin FAILED");
  
  // Start measuring
  ok= ccs811.start(CCS811_MODE_1SEC);
  if( !ok ) Serial.println("setup: CCS811 start FAILED");

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
}

void loop() {
  // Read
  uint16_t eco2, etvoc, errstat, raw;
  ccs811.read(&eco2,&etvoc,&errstat,&raw); 
  
  // Print measurement results based on status
  if( errstat==CCS811_ERRSTAT_OK ) { 
    Serial.print("CCS811: ");
    Serial.print("eco2=");  Serial.print(eco2);     Serial.print(" ppm  ");
    Serial.print("etvoc="); Serial.print(etvoc);    Serial.print(" ppb  ");
    Serial.println();
  } else if( errstat==CCS811_ERRSTAT_OK_NODATA ) {
    Serial.println("CCS811: waiting for (new) data");
  } else if( errstat & CCS811_ERRSTAT_I2CFAIL ) { 
    Serial.println("CCS811: I2C error");
  } else {
    Serial.print("CCS811: errstat="); Serial.print(errstat,HEX); 
    Serial.print("="); Serial.println( ccs811.errstat_str(errstat) ); 
  }
  
  // go to sleep for 1 minute
  Serial.println("Going to sleep now");
  delay(1000);
  Serial.flush(); 
//  delay(60000);
  esp_deep_sleep_start();
}`

And the result is continuously: "CCS811: waiting for (new) data"

If i use delay(60000) without sleep, it's all ok.

Could you please help me?

maarten-pennings, 2019-06-29:
Hi, I'm guessing you try to run this on ESP32. 
The `esp_deep_sleep_start()` puts the ESP32 in the deep sleep mode.

I'm not yet very familiar with the ESP32, and not at all with deep sleep, but when reading [randomnerdtutorials](https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources/), I found this quote

> In the `setup()` is where you should put your code. 
> In deep sleep, the sketch never reaches the `loop()` statement. 
> So, you need to write all the sketch in the `setup()`.

This makes sense to me, a deep sleep ends with a sort of "reset".

So, the bug seems to be your program and not in the CCS811 lib: when using deep sleep, all code needs to be in `setup` and needs to be prepared for the continues "resets".

apaczola, 2019-06-29:
Thank you very much, it seems that this has solved my problem.

maarten-pennings, 2019-06-29:
Glad it works.
If you're happy too, please close the issue.
Regards, Maarten

closed 2019-07-04 by apaczola