maarten-pennings/CCS811 #25: borken CJMU-8118

Aelphy, 2019-07-29:
I am trying to run your demo for CJMU-8118 on Chinese clone of Arduino Nano with old bootloader.
I checked that i2c works with monitor.
When I plug CJMU-8118 in it is not detected.
I do not use pull ups.
Could you suggest smth ?

maarten-pennings, 2019-07-29:
Have you connected it the same way as on my github home page?

Aelphy, 2019-07-29:
Yes. Now I managed to fix it. I found that rst pin should be 3.3V as well.
Thanks. You can close the issue, but it worth to mention somewhere that rst should be set to 3.3V.

maarten-pennings, 2019-07-29:
The CCS811 does not have an RST pin. 
So this seems board specific...

One of the users of my library is #bfaliszek, he seems to use the same board as you and he does not connect the RST pin:
https://github.com/bfaliszek/CJMCU-8118_InfluxDB/blob/master/README.md

Aelphy, 2019-07-29:
Yah, I have seen that. But got i2c errors. weird.
When I take rst pin off there is only i2c adress 0x40 available which stands for hdc1080.
When it is on - 0x5A becomes available as well.
However without rst pin hdc1080 it returns 125 degrees all the time and 0xFFFFFF as a serial number.

closed 2019-08-17 by maarten-pennings

Afollet, 2019-11-04:
Is there any update to this? I'm using the CJMU-8118 and I cannot get any reading from I2C

maarten-pennings, 2019-11-05:
From the comments above, it looks like the library is ok.

Do you have a RST pin (see posts above)?

Can you use a 3v3 device instead of the 5v Nano?

Can you confirm i2c messages are on the bus (eg have a working slave, or a logic analyzer)?

Aelphy, 2019-11-05:
It seems to be a problem of the board. I ordered two other CJMU-8218 and CJMU-8218. The first one didn't work at all the second one worked well. Also I noticed that they way how you solder the pins might matter (for my boards soldering should be done on the side with writings GND, SDA, etc).

Afollet, 2019-11-09:
Thanks for the quick replies.

I'm using a raspberry Pi to enumerate the i2c, but I don't get anything even at 3.3V. I tried setting the RST pin to 3.3V and that did not seem to work as well. 

I have observed that the connections on the board are a bit shotty, but I haven't got it to respond yet. I have not soldered the board to any wires. I am using a bread board. I am a bit of a biologist so I was hoping for a bit more plug'n'play with this chip.

Thanks

Aelphy, 2019-11-10:
Try to turn the board face down

maarten-pennings, 2019-11-17:
I never tried with a Raspberry Pi. Nice experiment.

> I tried setting the RST pin to 3.3V and that did not seem to work as well.
Good idea. The reset pin of the CCS811 is called nRESET, so indeed low active. Let's assume that the RST on your board is indeed also low-active.

I assume you did pull nWAKE low so that the CCS811 is alive and responding to I2C commands.