maarten-pennings/CCS811 #26: SetBaseline bounds

BorisDess, 2019-08-19:
Hello,

I played around with the set_baseline.
I don't know if this can be really considered as an issue but I wanted to share on it as it seems that the baseline if a very important parameter.

Unfortunately it seems that there's bounds to the value we can be setting.

When setting the baseline manually, the maximum value is 3327, the lower value is 173.
Don't asked me how I found those bounds ; trial and error for 30mn. 
When inputing a negative value, baseline is 65516,

maarten-pennings, 2019-08-21:
I'm not sure what you want to achieve. This chip deliberately keeps the baseline values "secret" so that we, as users, can not set a value. As the datasheet states: 
>BASELINE Register (0x11)
>A two byte read/write register which contains an encoded 
>version of the current baseline used in Algorithm Calculations. 
>A previously stored value may be written back to this two byte register
So, it is encoded, and we should only write back what we read.

BorisDess, 2019-08-21:
Actually I plan to retrieve the baseline to try to see if it can be useful in anyway when putting up the CCS811 against dedicated VOC sensors. I really need to get down to the manufacturer datasheet.

However yes indeed in no mean I want to set it myself.
I just observed that and thought it was worth sharing.

closed 2021-07-10 by maarten-pennings