maarten-pennings/CCS811 #27: Chinese module, I2C error

esp13, 2019-09-09:
Hi,

I found the firmware section in datasheet and was wondering if I already had the last one. So I founded your library.

But unfortunately like [https://github.com/maarten-pennings/CCS811/issues/6](url) and [https://github.com/maarten-pennings/CCS811/issues/7](url) I can't get your library working.

I can successfully use adafruit or sparkfun library ones. I don't touch wiring, just upload your code, and get this "I2C error":
```
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
ccs811: ping failed (VDD/GND connected? SDA/SCL connected?)
setup: CCS811 begin FAILED
setup: hardware    version: FFFFFFFF
setup: bootloader  version: FFFFFFFF
setup: application version: FFFFFFFF
setup: CCS811 start FAILED
CCS811: I2C error
CCS811: I2C error
CCS811: I2C error
CCS811: I2C error
```

I tried with and without setting I2C SDA/SCL pins (21,22)
I tried with and without ccs811.set_i2cdelay(50);
I tried with and without an WAK pin (17 / -1)
I tried with an without an I2C address set (0x5A)

I am on a ESP32 with a Chinese CSS811 module like [https://raw.githubusercontent.com/maarten-pennings/CCS811/master/wire-ccs811only.jpg](url)
I don't add resistors for pull-up.

Sorry for my poor English, hope this information help.

maarten-pennings, 2019-09-09:
Hmm
You tried a lot already.

The Arduino library for ESP32 version 1.0.0 has an i2c bug: it does not support repeated start conditions. This seems to be fixed in release 1.0.1. Which one are you using?

esp13, 2019-09-09:
Thank you for your answer

I was on 1.0.0 ...
I just upgraded to 1.0.2, but still the same :/

esp13, 2019-09-09:
Wait!
I tried to test again all combinations, and this one seems working:
```
CCS811 ccs811(-1);
Wire.begin(21,22); 
//ccs811.set_i2cdelay(50);
```

I do some tests more

esp13, 2019-09-09:
I get this:
```
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 1100
```

So I should upgrade my CSS811 firmware

esp13, 2019-09-09:
```
Starting CCS811 flasher
setup: library     version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
init: already has 2.0.0
```

Done!
Thank you much for your library, reposit and help :)

closed 2019-09-10 by esp13