maarten-pennings/CCS811 #28: nWake issue

hoangv, 2019-09-29:
Hi @maarten-pennings, thanks for this library, it is very helpful. I am having trouble getting nWAKE to work.  Do you have any suggestions to troubleshoot?

I can get ccs811basic working fine as long as I have nWake directly connected to GND.  The wiring is correct and matches the README.md for ESP8266.  Whenever I wire D3 to nWake and use `ccs811(D3)`, I get: `ccs811: ping failed (VDD/GND connected? SDA/SCL connected?)`.  I am using Wemos D1 mini with CJMCU-811.  I also tried using D4 pin instead of D3, and also tried flashing from 1.1 to 2.0, but still see the same error.

maarten-pennings, 2019-09-29:
Hi @hoangv, I'm a bit puzzled too, so allow me to ask some questions.

- you have an esp8266 and some ccs811 board
- you have 5 wires between the two boards: VDD to 3V3, GND to GND, SDA to D2, SCL to D1, and nWAKE to D3
- you run the standard ccs811basic.ino sketch without any modifications
- it does not work (you get ping failed)
- you change one wire (nWAKE to D3 into nWAKE to GND) and then it does work (no changes to the software)?

hoangv, 2019-09-30:
Hi @maarten-pennings, yes that was the situation yesterday.  This morning, I decided to start over with a new ESP8266 and rewired everything.  nWake appears to be working as expected.  I did not have a chance to troubleshoot further, but think it was either an issue with soldering, or with the ESP8266 module itself.  Thanks again.  Closing this out.

closed 2019-09-30 by hoangv