maarten-pennings/CCS811 #30: Measurements not realistic

HaViGit, 2019-12-28:
Hi Maarten,

First of all thanks for sharing this CCS811 info! I'm running your css811basic sketch on a Nano, WAKE connected to ground. The CO2 values in the serial monitor are not very realistic, mostly around 475 ppm while my Netatmo module in the same room reports about 1000 ppm. 

Info from sketch:
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 1100

The sensor is brand new and came from China (see image). Do I need more patient for a burn-in or do you have other suggestions?

Best regards,

Hans.

[image]

maarten-pennings, 2019-12-28:
Hi Hans,

That is a very open question.

The CCS811 does require to be "burned-in". This means that after around 1 week of run time the sensor is more stable. However, the internal controller knows about this burn in period and compensates during the first week. Especially firmware version 2.0.0 is improved here. You seem to have the older firmware, so you could consider upgrading (see the flash example).

Also, be aware that the CCS811 is a so-called MOX sensor. This class of sensors has a chemical reaction with the gases (an oxidation/reduction). But not all gasses react with the sensors. Most "Volatile organic compounds" react. This is the TVOC reading. But the sensor is insensitive for CO2. The trick is that the CO2 reading assumes the sensor is inside a building (iAQ - indoor air quality), and that humans are the (only) producer of CO2. So the gas sensor measures VOCs, assumes they are from humans, maps that to the amount of humans, and then maps that to the CO2 they would produce. But it can never be as accurate as a real CO2 sensor.

Also, the ccs811 is a relative sensor (that is the downside of such a cheap and small device). It senses of the air goes bad or improves; bit it will not give you an absolute value.

HaViGit, 2019-12-28:
Hi Maarten,

Thanks for your fast and clear answer! Not sure if the Netatmo Module data is correct but it is my only reference although they allways show high values.  I'll let the sensor burn in for a week or so and maybe do the upgrade thereafter.

Best regards, Hans.

closed 2019-12-28 by HaViGit