maarten-pennings/CCS811 #31: Thanks :)

fragsalat, 2020-01-08:
Hey Maarten,

I just wanted to thank you for this library. Now after I flashed my chip to the 2.1.0 version it gives quite nice results. As I'm currently learning rust I rewrote this library in rust with some minor changes. Also not everything is covered but the main functionality like reading data, setting baseline and environmental data and flashing the chip is supported and works. It was easy for me because of your work. That's why I just wanted to say thank you :)

If you want to have a look at my rust library check out https://crates.io/crates/ccs811

closed 2020-01-08 by fragsalat

maarten-pennings, 2020-01-08:
Hi Thomas,

Thanks for the compliments, and good work on the rust lib. You even have the flash function!

dadosch, 2020-04-10:
Hi @maarten-pennings 

I also want to say thanks. I just successfully upgraded the firmware to v2.

Stay healthy :)

maarten-pennings, 2020-04-13:
Thanks. I added you to the [success list](https://github.com/maarten-pennings/CCS811/blob/master/examples/ccs811flash/README.md#success-rate), but did know your hardware.