maarten-pennings/CCS811 #37: The library not work with ESP32

Dipieffe, 2020-03-12:
Hi i have prove this library on ESP32 with arduino but I have same error with I2C (I suppose) can you help me?

maarten-pennings, 2020-03-12:
I have tested it with esp32 as you can read on the homepage of this project. Your issue description does not give any information, so I don't know how to help you.

Which boards did you use. How did you connect. Which example did you run. What was the output?

Dipieffe, 2020-03-13:
First of all I want say thank you for the response. Have use this guide: https://github.com/maarten-pennings/CCS811 <https://github.com/maarten-pennings/CCS811>, with CCS811 and ESP32 devkitc v4 and I use Arduino. 

When I compile the code on Arduino the output is:
setup: Starting CCS811 basic demo
setup: ccs811 lib  version: 10
ccs811: ping failed (VDD/GND connected? SDA/SCL connected?)
setup: CCS811 begin FAILED
setup: hardware    version: FFFFFFFF
setup: bootloader  version: FFFFFFFF
setup: application version: FFFFFFFF
setup: CCS811 start FAILED
CCS811: I2C error

I have prove to connect nWAKE to GND and I have change the code in this mode: CCS811 ccs811(-1); but the result is the same.

maarten-pennings, 2020-03-14:
Hi, this helps. It means you have the basic app installed, and you have the lib installed.
You compiled them, and run them. But the app cannot connect to the hardware.

I suspect wrong wires, broken wires, or, worst case, a broken sensor.

Would it be possible to show me photo's of the wires. Where do the wires connect to the ESP, and where do they connect to the ccs811. Assuming you use wires of different colors, a zoomed-in  photo on the ESP side and a zoomed-in photo on the css811 side and a zoomed-out photo where everything is visible would be great.

Do you happen to have a logic analyser or scope? A great test is to put that on the pins of the CCS811 and check they have the correct signal.

coelner, 2020-04-04:
It could be another 'feature'. I have the same problem and I solved it by two things

1. use wire.begin with defined pins e.g. `Wire.begin(19, 23);` see #29 The ESP32 does not have specific I2C pins. 
2. hard reset of the vcc rail for the I2C sensors. The CCS811 somehow is a little bit picky about the initialisation.

closed 2021-07-10 by maarten-pennings