maarten-pennings/CCS811 #40: sensor missbehaviour

coelner, 2020-04-08:
It is not directly related to this library itself, but maybe the right sport to discuss this:
```
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000
[...]
148:	Temp: 29.51°C	Humidity: 23.67% RH	Pressure: 1001.30hPa	eCO2: 1726 ppm 	eTVOC: 366 ppb 	
148:	Temp: 29.75°C	Humidity: 24.82% RH	Pressure: 1001.30hPa	eCO2: 1625 ppm 	eTVOC: 268 ppb 	
baseline for this sensor: 0xEBB0
148:	Temp: 29.88°C	Humidity: 26.04% RH	Pressure: 1001.29hPa	eCO2: 1621 ppm 	eTVOC: 265 ppb 	
149:	Temp: 30.03°C	Humidity: 24.51% RH	Pressure: 1001.22hPa	eCO2: 1692 ppm 	eTVOC: 330 ppb 	
149:	Temp: 30.04°C	Humidity: 24.36% RH	Pressure: 1001.32hPa	eCO2: 1702 ppm 	eTVOC: 341 ppb 	
149:	Temp: 29.79°C	Humidity: 24.80% RH	Pressure: 1001.28hPa	eCO2: 782 ppm 	eTVOC: 57 ppb 	
149:	Temp: 30.04°C	Humidity: 23.65% RH	Pressure: 1001.32hPa	eCO2: 570 ppm 	eTVOC: 25 ppb 	
149:	Temp: 29.65°C	Humidity: 25.73% RH	Pressure: 1001.29hPa	eCO2: 423 ppm 	eTVOC: 3 ppb 	
149:	Temp: 29.66°C	Humidity: 24.65% RH	Pressure: 1001.22hPa	eCO2: 403 ppm 	eTVOC: 0 ppb 	
149:	Temp: 29.69°C	Humidity: 25.42% RH	Pressure: 1001.23hPa	eCO2: 405 ppm 	eTVOC: 0 ppb 	
149:	Temp: 29.65°C	Humidity: 23.97% RH	Pressure: 1001.22hPa	eCO2: 415 ppm 	eTVOC: 2 ppb 	
149:	Temp: 29.70°C	Humidity: 23.55% RH	Pressure: 1001.26hPa	eCO2: 402 ppm 	eTVOC: 0 ppb 	
149:	Temp: 29.47°C	Humidity: 23.88% RH	Pressure: 1001.30hPa	eCO2: 401 ppm 	eTVOC: 0 ppb 	
150:	Temp: 29.31°C	Humidity: 23.96% RH	Pressure: 1001.21hPa	eCO2: 403 ppm 	eTVOC: 0 ppb 	
150:	Temp: 28.56°C	Humidity: 24.77% RH	Pressure: 1001.23hPa	eCO2: 408 ppm 	eTVOC: 1 ppb 	
150:	Temp: 28.93°C	Humidity: 24.37% RH	Pressure: 1001.27hPa	eCO2: 412 ppm 	eTVOC: 1 ppb 	
150:	Temp: 28.56°C	Humidity: 24.88% RH	Pressure: 1001.25hPa	eCO2: 409 ppm 	eTVOC: 1 ppb 	
150:	Temp: 28.32°C	Humidity: 25.19% RH	Pressure: 1001.30hPa	CCS811: errstat=899=--vhXmrwF--AD-iE
150:	Temp: 28.70°C	Humidity: 25.07% RH	Pressure: 1001.18hPa	CCS811: errstat=899=--vhXmrwF--AD-iE
150:	Temp: 28.63°C	Humidity: 25.25% RH	Pressure: 1001.18hPa	eCO2: 466 ppm 	eTVOC: 9 ppb 	
baseline for this sensor: 0x9DAD
150:	Temp: 28.76°C	Humidity: 25.17% RH	Pressure: 1001.27hPa	CCS811: errstat=899=--vhXmrwF--AD-iE
150:	Temp: 28.68°C	Humidity: 30.86% RH	Pressure: 1001.32hPa	eCO2: 514 ppm 	eTVOC: 25 ppb 	
150:	Temp: 28.38°C	Humidity: 32.08% RH	Pressure: 1001.28hPa	eCO2: 1961 ppm 	eTVOC: 756 ppb 	
151:	Temp: 27.83°C	Humidity: 32.73% RH	Pressure: 1001.28hPa	eCO2: 2866 ppm 	eTVOC: 2566 ppb 	
151:	Temp: 27.45°C	Humidity: 33.58% RH	Pressure: 1001.31hPa	eCO2: 3702 ppm 	eTVOC: 4359 ppb 
151:	Temp: 27.07°C	Humidity: 34.13% RH	Pressure: 1001.28hPa	eCO2: 4463 ppm 	eTVOC: 5976 ppb 	
151:	Temp: 26.83°C	Humidity: 34.89% RH	Pressure: 1001.30hPa	eCO2: 5092 ppm 	eTVOC: 7319 ppb 	
151:	Temp: 26.56°C	Humidity: 35.67% RH	Pressure: 1001.24hPa	eCO2: 5750 ppm 	eTVOC: 8736 ppb 	
151:	Temp: 26.37°C	Humidity: 36.19% RH	Pressure: 1001.26hPa	eCO2: 6383 ppm 	eTVOC: 10079 ppb 	
151:	Temp: 26.20°C	Humidity: 36.54% RH	Pressure: 1001.21hPa	eCO2: 6940 ppm 	eTVOC: 11271 ppb 	
151:	Temp: 26.09°C	Humidity: 36.84% RH	Pressure: 1001.26hPa	eCO2: 7859 ppm 	eTVOC: 13267 ppb 	
151:	Temp: 25.99°C	Humidity: 36.72% RH	Pressure: 1001.26hPa	eCO2: 8421 ppm 	eTVOC: 14433 ppb 	
```
I put the sensor out of the window to get the baseline. The measurement is done every second but the values get summed up for a minute. First of all, there occurs some errors because of maximum resistance was reached. After this errors the sensors was placed inside. The absolute values are far beyond useful. 
The sensor itself is burned-in (at least for a week) and this specific print-out was done after three hours running.

coelner, 2020-04-09:
I have a SGP30 and CCS811 at the same spot and did some graphs. The sensor is on the window board and the bay is separated with a paper curtain from the room itself. The curtail has got a gap  to the board about 3cm. 
Measurement is every second.
 - 300s I opened a windows
[image]
[image]
[image]
[image]
After 90 minutes it looks like this:
[image]
 - 1566 I blow into the gap
 - 1600-1650 I wiggle the curtain a little bit to simulate some air flow
[image]

maarten-pennings, 2020-04-09:
Nice graphs. 

I agree that values like `eCO2: 8421 ppm eTVOC: 14433 ppb` make no sense.
I also cannot explain the differences between CCS811 and SGP30. The `CCS811: errstat=899=--vhXmrwF--AD-iE` does not raise my trust in the values.

Please do note that these sensors are TVOC sensors *not* CO2 sensors (the CO2 is computed from the TVOC, assuming the sensor is indoor).

coelner, 2020-04-10:
The SGP30 (I use the sparfun library) is also not very trustworthy, sometimes it keeps the eCO2 value at 400 which can't be true. (even for a MOX sensor, which should change the resistance by blowing over it).

maarten-pennings, 2020-04-13:
These are _relative_ sensors.

If you take a bottle, spray a bit of alcohol in it, hang your sensor in it, close the bottle, and then switch on the sensor, it will happily report clean air (assuming you do not use the set baseline feature).

It also means that the 'zero' value is arbitrary. what the sensors typically do is take the value at power on (again, assuming you do not use the set baseline feature), and use that as reference. Typically assuming clean air, and mapping that to 400 (SGP30), 450 (CCS811), 500 (iAQcore).

By the way, I heard once, blowing on the sensor is a bit abusive. I forgot why. Maybe you cool the heater too much and it needs to step current, maybe it was the moist in your breath. Anyhow they told me not to blow, but to use a marker pen or breath some breath freshener on it.

Good luck

closed 2021-07-10 by maarten-pennings