maarten-pennings/CCS811 #41: platformio: change name

dadosch, 2020-04-10:
Apparently, there are two libraries on platformio with the same name "CCS811":

Yours https://platformio.org/lib/show/1609/CCS811
and a other one. https://platformio.org/lib/show/3333/CCS811

If you specify the library by name, it loads the other one. I'm not sure if this is a issue of platformio or your library.

maarten-pennings, 2020-04-13:
Hi @dadosch,

Uhh, what can I say. I didn't even know I was on Platform IO! I did not put my library there.

The lib is called the same as the sensor name. Sounds like a logical choice. And Adafruit made the same choice. And apparently platformio allows two libs with the same name, that is good. I guess that it then also has a mechanism to select another one if you don't like the default one. I never worked with platformio, so I can't help you here.

I just that a bit of quick clicking in platformio and found [something](https://platformio.org/lib/show/1609/CCS811/installation) that might help:

```
# Using library Id
 platformio lib install 1609

# Using library Name
 platformio lib install "CCS811"

# Install specific version
 platformio lib install 1609@10.0.0
```

Thanks for the tip - I feel a bit proud now that the copied my work.

dadosch, 2020-04-13:
I found your library through platformio, so I actually tried to install your lib with 

` platformio lib install "CCS811"`

but it installed https://platformio.org/lib/show/3333/CCS811 instead.

Platformio is taking this name from your https://github.com/maarten-pennings/CCS811/blob/master/library.properties file. Adafruit has the following name: `name=Adafruit CCS811 Library`

closed 2021-07-10 by maarten-pennings