maarten-pennings/CCS811 #43: set_envdata() t & h format?

lloydrichards, 2020-05-03:
I've been making a bit of a sensor matrix and have included the CCS811 along with a slew of other environmental sensors.  After a couple revisions and different libraries, I've found yours to the most robust for testing my hardware.  While I'm getting consistent readings from the sensor now, I'm looking more at calibrating the results and so look to set the environmental information via the set_envdata().  I've tried just feeding in the float values from the SHT31 and BMP280 that I have onboard but the readings I get back are very skewed (~6500ppm).  Have been trying to see If this is a burn-in issue but I think more likely its a formatting issue between what the set_envdat() needs.  

Does it work with just any float or does there need to be some additional calculations or rounding in order to feed the values into the functions?

maarten-pennings, 2020-05-03:
Hi, my ccs811 library has two set_envdata methods. I quote the header file:

```
bool set_envdata(uint16_t t, uint16_t h);                                 // Writes t and h to ENV_DATA (see datasheet for format). Returns false on I2C problems.
bool set_envdata210(uint16_t t, uint16_t h);                              // Writes t and h (in ENS210 format) to ENV_DATA. Returns false on I2C problems.
```
So the plain `set_envdata()` uses a format native to the chip, and you need to look that up in the datasheet. 

The other one `set_envdata210` uses the format from the ens210, which is a temperature humidity chip from the same manufacturer.

There is no general `set_envdata` using floats. I dislike float for small embedded microcontrollers :-(

lloydrichards, 2020-05-04:
Having a look at the datasheet it says `Temperature is stored as an unsigned 16 bits integer in 1/512 degrees`. Will need to do some conversion I guess.  If I figure this out I'll post up the results incase someone else runs into the same problem

maarten-pennings, 2020-05-04:
It also says "there is an offset: 0 maps to -25°C."
So, this is my guess; didn't test it!

```
float TinC;
unsigned int Tfor811 = (TinC+25) * 512;
```

lloydrichards, 2020-05-04:
Have played around in the code and did a bit of research into what Adafruit and SparkFun have done and come up with the following chunk:

```
    double avgTemp = (temp1 + temp2 + temp3) / 3;
    int avgHum = (hum1 + hum2) / 2;

    float fract = modf(avgTemp, &avgTemp);
    uint16_t tempHIGH = (((uint16_t)avgTemp + 25) << 9);
    uint16_t tempLOW = ((uint16_t)(fract / 0.001953125) & 0x1FF);

    uint16_t tempCONVERT = (tempHIGH | tempLOW);
    uint16_t humCONVERT = avgHum << 1 | 0x00;

    ccs811.set_envdata(tempCONVERT, humCONVERT);
```

this could very well be written as a function if someone was so inclined, but for my purposed doing the quick conversion before was all I needed :)

Thanks for the help Maarten, great work on the library :D

maarten-pennings, 2020-05-04:
I believe your code is NOT correct. For negative numbers, the fraction (after modf) is negative!
To fix that, add the 25 before modf.

Secondly, the code is obscuring that you do two times the same (namely *512 is first written as <<9 and then as /0.001953125).

Note that 0≤fract<1, so 0≤fract*512<512 so the &0x1ff is not needed.

So I think you need this.

```
 fract = modf(avgTemp+25, &intg);
uint16_t tempHIGH = (uint16_t)(intg*512);
uint16_t tempLOW = (uint16_t)(fract*512);
uint16_t tempCONVERT = (tempHIGH | tempLOW);
```

Did you check ... I think this gives the same as

```
uint16_t tempCONVERT = (avgTemp+25) * 512;
```

If this is not the same, please let me know with an example.

janjagusch, 2021-01-12:
Maybe useful for somebody in the future. I had similar issues when trying to compensate for the environment using the DHT22 with sudden eCO2 values of > 1000. Turns out the DHT22 sensor sends some `null` values for temperature and humidity, before sending correct data. Calculating the compensation from `null` results in a very large number, which ultimately messes up the readings of the CCS811. By making sure temperature and humidity are not `null` before setting the environment data, I was able to get correct (or at least correct looking results).

```cpp
uint16_t humidity_to_compensation(float humidity) {
  // Humidity is stored as an unsigned 16 bits in 1/512%RH.
  // The default value is 50% = 0x64, 0x00.
  // As an example 48.5% humidity would be 0x61, 0x00.
  // return uint16(50 * 512); // Default value
  return uint16(humidity * 512);
}

uint16_t temperature_to_compensation(float temperature) {
  // Temperature is stored as an unsigned 16 bits integer in 1/512 degrees;
  // there is an offset: 0 maps to -25°C. The default value is 25°C = 0x64, 0x00.
  // As an example 23.5% temperature would be 0x61, 0x00.
  // return uint16(50 * 512); // Default value
  return uint16((temperature + 25) * 512);
}

void compensate_for_environment() {
  float temperature, humidity;
  temperature = dht.readTemperature();
  humidity = dht.readHumidity();
  // Make sure temperature and humidity are set,
  // otherwise CCS sensor values will be all over the place.
  if (!(isnan(temperature) | isnan(humidity))) {
      ccs811.set_envdata(temperature_to_compensation(temperature), humidity_to_compensation(humidity));
  }
}
```

maarten-pennings, 2021-07-10:
I leave this issue open; the functions look good, but I didn't implement them yet.

redstorm1, 2021-07-13:
> 
> ```c++
> uint16_t humidity_to_compensation(float humidity) {
>   // Humidity is stored as an unsigned 16 bits in 1/512%RH.
>   // The default value is 50% = 0x64, 0x00.
>   // As an example 48.5% humidity would be 0x61, 0x00.
>   // return uint16(50 * 512); // Default value
>   return uint16(humidity * 512);
> }
> 
> uint16_t temperature_to_compensation(float temperature) {
>   // Temperature is stored as an unsigned 16 bits integer in 1/512 degrees;
>   // there is an offset: 0 maps to -25°C. The default value is 25°C = 0x64, 0x00.
>   // As an example 23.5% temperature would be 0x61, 0x00.
>   // return uint16(50 * 512); // Default value
>   return uint16((temperature + 25) * 512);
> }
> 
> void compensate_for_environment() {
>   float temperature, humidity;
>   temperature = dht.readTemperature();
>   humidity = dht.readHumidity();
>   // Make sure temperature and humidity are set,
>   // otherwise CCS sensor values will be all over the place.
>   if (!(isnan(temperature) | isnan(humidity))) {
>       ccs811.set_envdata(temperature_to_compensation(temperature), humidity_to_compensation(humidity));
>   }
> }
> ```

Thanks for this, i am using a DTH22 also and was looking for the right conversion to set the environment data. Readings looking good. Thanks for posting. maybe move the conversion to the right format into the CSS library so only have to pass the correct temp and humidity values to make it easy for anyone else in the future.

redstorm1, 2021-07-13:
I double checked the conversions and I agree with janjagusch
return uint16(50 * 512); // Default value 0x64
return uint16((25+25) * 512); // Default value 0x64

Merge it.

maarten-pennings, 2021-07-18:
Merged

closed 2021-07-18 by maarten-pennings