maarten-pennings/CCS811 #44: CJMCU-8118 ; 3 degrees over and 5% humity bellow.

woiptcs, 2020-05-03:
Hello Marteen , pleased to contact you.

We've been fighing for weeks to get an accurate temp/humity/CO2/TVOC measurements from a CJMCU-8118, using ESP-IDF.

Best results have been 2ºC over , and 5% less humity than measured with 3 other instruments.

If I use modes 0x10 or 0x40 on the CCS811 , it seems the heater (all time on) is affecting the measurement on the HDC1080. (2-3ºC over , 5-7%Hum below)

If I use mode 0x20 or 0x30 in CCS811 , HDC1080 temperature and humidity give good valuea but I'm unable to get correct measurements from the CSS811. (allways same wrong value , or 0x400, 0)..

If we use the INT pin , in order to get the value from the sensor when it's ready you think this will fix the issue? 

We are using a self delevolped application written entirely in ESP-IDF.

On other hand, our sensor hardware version  is 1.2 and app version is 1.1 when we try to upgrade to app version 2.00 or 2.01 we can not get any measurements again.

Do you think which is the minimum hardware version we need to upgrade app to 2.x? 

Thank you in advance and congrats for your great work .-

CCSbasic log : 

18:48:38.027 -> setup: Starting CCS811 basic demo
18:48:38.027 -> setup: ccs811 lib  version: 10
18:48:38.027 -> setup: hardware    version: 12
18:48:38.027 -> setup: bootloader  version: 1000
18:48:38.027 -> setup: application version: 1100
18:48:38.027 -> CCS811: waiting for (new) data
18:48:39.022 -> CCS811: waiting for (new) data
18:48:40.016 -> CCS811: eco2=0 ppm  etvoc=0 ppb  
18:48:41.044 -> CCS811: eco2=0 ppm  etvoc=0 ppb  
18:48:42.039 -> CCS811: eco2=0 ppm  etvoc=0 ppb  
18:48:43.033 -> CCS811: eco2=400 ppm  etvoc=0 ppb  
18:48:44.028 -> CCS811: eco2=405 ppm  etvoc=0 ppb  
18:48:45.056 -> CCS811: eco2=405 ppm  etvoc=0 ppb  
18:48:46.050 -> CCS811: eco2=400 ppm  etvoc=0 ppb  
18:48:47.045 -> CCS811: eco2=400 ppm  etvoc=0 ppb  
18:48:48.039 -> CCS811: eco2=400 ppm  etvoc=0 ppb  
18:48:49.034 -> CCS811: eco2=400 ppm  etvoc=0 ppb  

AIDomotics app log : 

I (602) AIDOM-SENSOR System Log: ..::AIDomotics::.. - T,H,CO2,TVOC Sensor
I (612) AIDOM-SENSOR System Log: Developed by: Santiago Moncada, Ruben Montferrer, Andres Prieto
I (622) AIDOM-SENSOR System Log: Coded For: UPC University and the whole world
I (622) AIDOM-SENSOR System Log: WARRANTY: This software have NOT any warranty
I (632) AIDOM-SENSOR System Log: Executing POST...
Turning on the R LED
Turning off the R LED
Turning on the G LED
Turning off the G LED
Turning on the B LED
Turning off the B LED
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
I (1072) AIDOM-SENSOR System Log: POST Completed...
I (1072) AIDOM-SENSOR System Log: COLD START DETECTED ...
I (1082) AIDOM-SENSOR System Log: LED_EVENT_RECEIVED
I (1082) AIDOM-SENSOR System Log: LED REQUEST = 12
I (1092) LED CONTROL: LED ACTIVE STATUS IS: 12
I (1122) 802.11: ESP_WIFI_MODE_STA
I (1132) wifi: wifi driver task: 3ffc4c18, prio:23, stack:3584, core=0
I (1132) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1132) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
Task List : 
ipc1           	R	24	564	3	1
main_loop      	R	24	2492	14	0
main           	R	1	404	5	0
IDLE1          	R	0	1116	7	1
IDLE0          	R	0	1088	6	0
button_handler 	B	10	1616	13	0
led_active     	B	10	520	15	-1
Tmr Svc        	B	1	1596	8	0
esp_timer      	B	22	3640	1	0
aidom_events_ta	B	10	2160	12	1
ipc0           	B	24	572	2	0
I (1162) wifi: wifi firmware version: a44d1c6
I (1182) wifi: config NVS flash: enabled
I (1192) wifi: config nano formating: disabled
I (1192) wifi: Init dynamic tx buffer num: 32
I (1202) wifi: Init data frame dynamic rx buffer num: 32
I (1202) wifi: Init management frame dynamic rx buffer num: 32
I (1212) wifi: Init management short buffer num: 32
I (1212) wifi: Init static rx buffer size: 1600
I (1212) wifi: Init static rx buffer num: 10
I (1222) wifi: Init dynamic rx buffer num: 32
I (1312) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1312) wifi: mode : sta (a4:cf:12:32:f7:8c)
I (1322) 802.11: wifi_init_sta finished.
I (1322) 802.11: connect to ap SSID:AIDOMGW51 password: XXXXX 
I (1322) AIDOM SENSING: ENPOWERING SENSORS ....
I (1432) AIDOM SENSING: CONFIGURING SENSORS ...
I (1442) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1442) wifi: state: init -> auth (b0)
I (1442) wifi: state: auth -> assoc (0)
I (1452) wifi: state: assoc -> run (10)
Current Sensor Status :W (1452) AIDOM SENSING: ESP_OK
0x10 0x00 
I (1472) wifi: connected with AIDOMGW51, aid = 3, channel 11, BW20, bssid = 00:e0:4c:2b:3e:00, security type: 3, phy: bgn, rssi: -51
Current measuring mode :W (1472) AIDOM SENSING: ESP_OK
0x20 
I (1472) AIDOM SENSING: WARMING SENSORS ...
I (1482) wifi: pm start, type: 1

I (1482) wifi: AP's beacon interval = 102400 us, DTIM period = 1
I (1532) AIDOM SENSING: READING MEASUREMENTS ...
I (1652) AIDOM SENSING: T/H/Co2/TVOC : 28.10Cº, 	 49.27, 	 477ppms,	11ppms
I (3122) esp_netif_handlers: sta ip: 10.254.254.12, mask: 255.255.255.0, gw: 10.254.254.1
I (3122) power_save: got ip: 10.254.254.12
JSON len = 79  
JSON Payload ={"temperature":28.0987548828125,"humidity":49.273681640625,"co2":477,"tvoc":11} 
I (4322) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4332) MQTTS_FUNCTIONS: Other event id:7
I (4342) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (4352) MQTTS_FUNCTIONS: MQTT_EVENT_CONNECTED
I (5332) MQTTS_FUNCTIONS: MQTT Topic : dt/device/a4:cf:12:32:f7:8c/sensor
I (5332) MQTTS_FUNCTIONS: MQTT Data : {"temperature":28.0987548828125,"humidity":49.273681640625,"co2":477,"tvoc":11}
I (5332) MQTTS_FUNCTIONS: Publishing data...
I (5342) MQTTS_FUNCTIONS: Sent publish successful, msg_id=31063
I (5342) AIDOM-SENSOR System Log: LED_EVENT_RECEIVED
I (5352) MQTTS_FUNCTIONS: MQTT_EVENT_PUBLISHED, msg_id=31063
I (5352) AIDOM-SENSOR System Log: LED REQUEST = 33
I (5412) LED CONTROL: LED ACTIVE STATUS IS: 33
I (5842) AIDOM-SENSOR System Log: LED_EVENT_RECEIVED
I (5842) AIDOM-SENSOR System Log: LED REQUEST = 0
I (5902) LED CONTROL: LED ACTIVE STATUS IS: 0
I (5942) AIDOM-SENSOR System Log: Enabling timer wakeup, 60 seconds
I (5962) AIDOM-SENSOR System Log: Entering deep sleep....
I (5962) wifi: state: run -> init (0)
I (5962) wifi: pm stop, total sleep time: 3127797 us / 4475512 us

I (5962) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
E (5972) wifi: esp_wifi_connect 1148 wifi not start
I (5982) wifi: flush txq
I (5982) wifi: stop sw txq
I (5982) wifi: lmac stop hw txq

Done

Santiago .-

nordicp, 2020-05-07:
The rh/T sensor on this little board is pretty close to the CCS811 which generates some heat. The PCB is small and the trade off is that the gas sensor influences T and rh reading, the +2 to +3C sounds reasonable,  as a result rh is lower than measured at with a seperate device running at lower (ambient) temperature.
I flashed a sensor from 1.0.0 to 2.0.0 using a ProMini 3.3V and now will give it a try with ESP32. Since ESP32's IC clockstretchig is fixed and cannot be set I wonder if this causes problems.

nordicp, 2020-05-07:
Just noticed that #39 reported an issue with ESP32 and running the ESP32 at lower frequency solved the problem.

maarten-pennings, 2020-05-10:
> On other hand, our sensor hardware version is 1.2 and app version is 1.1 when we try to upgrade to app version 2.00 or 2.01 we can not get any measurements again.

What do you mean? You have multiple CCS811s and you upgraded some, and then they don't work anymore?

> Do you think which is the minimum hardware version we need to upgrade app to 2.x?
To the best of my knowledge there is only one hardware version.

woiptcs, 2020-05-11:
Thanks for your answers Maarten and Nordicp !, 

After all research I think , the proximity of both sensors is the key of the wrong measures.
Regarding to the APP version upgrade , when I upgrade the sensor to V.2.00 or 2.01 , our application does not work again, also using Arduino examples , we need to do some more research again.

Do you have the v1.0 or v.1.1 binaries in order to downgrade the current sensors? 

Thank you in advance ! 

Santiago .-

maarten-pennings, 2020-05-13:
No sorry, when I wrote this library, the manufacturer only published these new binaries, not the old one.

woiptcs, 2020-05-13:
Ok Thanks anyway Marteen , good work and stay home brother.

va escriure:

maarten-pennings, 2021-07-10:
Somebody sent my older versions; there were added.

closed 2021-07-10 by maarten-pennings