maarten-pennings/CCS811 #47: How To: use multiple CJMCU-8110

KapakoProf, 2020-05-14:
Hi, I am rather new in programming and wonder how to compact the code I wrote for using a flexible number of CJMCU-8110 connected to my NodeMCU.
In the attached test code below I have 2 connected, but in the near future I want to expand to 3 or 4 and would like to drive that by a variable (f.i. int SensorNumber = 2) instead of duplicating the code for each sensor.

Who can guide me to a solution?

[20200514_PostedCode.txt](https://github.com/maarten-pennings/CCS811/files/4627006/20200514_PostedCode.txt)

maarten-pennings, 2020-05-14:
There are two problems.
The first problem, the one you ask, is a software design problem "how to avoid duplication of code". What I would suggest is to create an _array_ of ccs811 objects. That way you can simply loop over them.

The second problem is a hardware problem. I2C devices have an address. On one I2C bus every address can only occur once! The ccs811 helps a bit: there is an io pin you can pull high or low and with that you can select between two addresses. In short one I2C bus has max two ccs811s.

KapakoProf, 2020-05-14:
Thanks Maarten. I have changed my code to use an array and now loop the code which is single available.
While I do not have enough pins available (other sensors and muxes) yo create multiple I2C busses I now share the SCL pin between 4 css811 and have 2 different pins for SDA which are set in the loop. Works perfect.
May be I eventually will end up in using multiple NodeMCU, but for now it is ok.
Thanks.

closed 2020-05-14 by KapakoProf