maarten-pennings/CCS811 #49: Downgrade to CCS811 Old Firmware v1.0.0

daferdur, 2020-05-16:
Hi guys!
I've modified my chinesse CSS811 board and soldered a 10k SMD NTC and a 10k SMD Resitor (R3 and R4 in original schematic) in orther to get Temperature readings.
The fact was that I had this board updated to firmware version 2.0.1 and it wasn't impossible to get temperature readings with this firmware.

So after a lot of searching Internet, I've found the old firmware to get temperature readings. 
And of course downgraded it with success.

Here you have that firmware.
[ccs811_fw_app_1v0p0.bin.zip](https://github.com/maarten-pennings/CCS811/files/4638037/ccs811_fw_app_1v0p0.bin.zip)

Hope it will serve to somebody.

maarten-pennings, 2020-05-17:
Thanks. I'll add it to the repo. Where did find it?

daferdur, 2020-05-17:
https://github.com/ryankurte/efm32-base/blob/master/hardware/kit/common/bsp/thunderboard/ccs811_fw_app_1v0p0.bin

daferdur, 2020-05-17:
[image]

Here you can see R3 qnd R4

daferdur, 2020-05-17:
[image]

And here my updated CSS811 board.

maarten-pennings, 2020-05-17:
Nice! 

I didn't realize the board had the mounting pads already, 

Thanks for the 1.0.0 bin file; I have added it to the repo.

I've also changed the python script. The old one was still working for me, so I'm not sure what was the problem for you. It now uses f-strings.

daferdur, 2020-05-18:
Thanks to you for your contribution.

closed 2020-05-18 by maarten-pennings