maarten-pennings/CCS811 #57: Having a simple questions about the CCS811 function when it comes to modes

servercookie, 2020-10-18:
@maarten-pennings 

I apologize if this questions have been asked before, but afterchecking around i did not find any info about this in the other posts.

So we are using the CCS811 in a indoor sensor device, CCS811 is running the latest 2.0.1 version of the firmware. 
When running it in constantly in Mode 3 it seems to behave quite well, after the initial burn in the readings seems to make sense. TVOC = 240 with minor fluxuations and eCO2 = in the 1600-1800...cant recal the excat value now.

Since the device is battery driven we tried to save some power by doing the following: 

1: Let device run for 30min in mode 3 so that readings are stable, documentation says 20min
After 30min we take a reading, then it goes back to Idle Mode, but doing this we see some odd behavior

If we use this 30min interval were we switch from 30min runtime and back to idle, the reading shows tvoc = 0 and eCo2 = 400 
And its basically stuck there, meaning values dont move at all unless we beath on it or leave a open pen next to the sensor, then we get a spike. I know we have good air quality in the office, but its not that great:) 

Are we doing it wrong or is there another way to do this, or should we just forget about this and just let it run in mode 3 the entire time ?

Any ideas and suggestions would be highly appritiated 

Best regards
SC Naimy

servercookie, 2020-10-19:
I also found another issue while running 2.0.1, when you change mode from Mode 3 to Mode 0, this somehow reset the baseline. 

This is really odd , it does not say anywere in the documentation to save baseline before changing modes, we only switch between mode 0 and mode 3. 

Any insight would be helpful

maarten-pennings, 2020-10-19:
As the CCS811 datasheet explains, baseline management is described in AN000370. There you find your answer:

> If multiple IAQ operating modes are used on the same sensor, the baseline should be stored for 
each mode. The baseline is mode specific, it is therefore not possible to use the same baseline 
value for all modes.

servercookie, 2020-10-20:
@maarten-pennings 

Thank you for responsing, so basically it does not store the baseline when going from Mode 3 to 0. 
So we have to read,save, restore when switching between these modes. 

On final question that might sounds silly, when staying is Mode 0..i assume it does not count as burn in ? 
It would need to stay either mode 1-3 to be counted as burn in time right ? 

Thanks buddy

maarten-pennings, 2020-10-20:
You have to save the baseline _per mode_.

Burn-in is the process of stabilizing the sensor after manufacturing; the device needs to settle itself, getting used to the high temperatures of the heater. I would expect that only the _amount of hours at high temperature_ matters (ie heater on), not in which mode this takes place (of course the modes differ in the time the heater is on).

servercookie, 2020-10-20:
@maarten-pennings  many thanks again for you help 

We have started implementation of the  save/restore baseline function as the documention tells us to do. 
Thank you for clearing this up for us :)

closed 2021-07-10 by maarten-pennings