maarten-pennings/CCS811 #59: [Question] Modes, Burn-In - Several Questions

DerPicknicker, 2020-10-27:
Good afternoon, 

I have discovered your library and read your posts and already learned a lot...Thank you for your detailed insights. 
But I still have some (probably simple questions for you).

I have already looked into the datasheet and found the modes there. I have a few questions: 

- Does your library use the modes? 
- If so, how can I set the modes? 
- Can the sensor sleep for 5 minutes or must it be permanently active? 
- Are the values inaccurate if it has not measured anything for too long? 

Do you have to use firmware 2.0.x to get useful results or is it enough to burn the sensor with firmware 1.0 for 48 hours? 

Are there any tips under which conditions the burn-in has to take place? 

Thank you for Reading my Post.

Best Regards!

DerPicknicker, 2020-10-27:
EDIT: 

Now I can answer some of my Question : 
- Does your library use the modes?
- If so, how can I set the modes?

In this Line you set the Mode, this must happen in the Setup() Function. My other Questions aren’t answered yet. I hope @maarten-pennings have some time to answer the other ones :)

`ccs811.start(CCS811_MODE_1SEC); // CCS811_MODE_IDLE (Mode0); CCS811_MODE_1SEC (Mode1); CCS811_MODE_10SEC (Mode2); CCS811_MODE_60SEC (Mode3)`

maarten-pennings, 2020-10-31:
> Can the sensor sleep for 5 minutes or must it be permanently active?

As the datasheet explains: _After writing to MEAS_MODE to configure the sensor in mode 1-4, run CCS811 for 20 minutes, before accurate readings are generated._ 

So, each time you switch mode (which you must if the device is powered on) you have to condition the device for 20 min. For lowest power, put the device in 60s mode.

> Are the values inaccurate if it has not measured anything for too long?

What do you mean with "has not measured"?

Firstly, the device must be running at least 20 min before it is conditioned.
Secondly, the reading that you get is relative to the cleanest air the device has seen in the last 24hr. See "Automatic Baseline Correction" in the datasheet.

DerPicknicker, 2020-11-05:
After some time the Results of the measurements are totally weird... Overnight I measured 1k-4k ppm. The Window was opened at the Sensor was 20cm away from the Window... I ordered a new One to check if my first Sensor was defect. And if not I will test it again with an BME280 for the Temperature Compensation.

closed 2021-07-10 by maarten-pennings