maarten-pennings/CCS811 #60: ESP32 ERRO I2C

juka-lp, 2020-11-03:
I'm having a problem with the ccs811, and for days I can't find the reason.
When I run CCS811 Basic, I get the error "CCS811: I2C error"
And even if I try to update the firmware, I get the error:

Starting CCS811 flasher
setup: library version: 10
ccs811: ping failed (VDD / GND connected? SDA / SCL connected?)
setup: CCS811 begin FAILED
setup: hardware version: FFFFFFFF
setup: bootloader version: FFFFFFFF
setup: application version: FFFFFFFF
setup: starting flash of 'CCS811_FW_App_v2-0-0.bin' in 5 seconds

I've tried everything, even signaling the pin on Wire, does anyone have any idea what to do?

maarten-pennings, 2020-11-03:
Hi, the line
```
ccs811: ping failed (VDD / GND connected? SDA / SCL connected?)
```
is important. It is the very first time the library tries to communicate with the ccs811. The communication is the lightest possible, a ping. And that fails.

This means you have a hardware problem:
 - ccs811 is broken
 - the i2c block of your micro is broken
 - the data connection (sda and scl lines) is wrong (pullups?)
 - the power connection is wrong (vcc, gnd)
 - power mismatch (5v vs 3v3)
 - nwake is not grounded (simple setup)

Flashing a new firmware is useless, you first need to establish communication.

juka-lp, 2020-11-03:
The pins are connected in the right place, one of the times I tried everything went right, but out of nowhere it went back to this error forever.
I run an i2c scanner, and the i2c address appears, 0x16 is the most frequent

closed 2021-07-10 by maarten-pennings