maarten-pennings/CCS811 #61: IC2 Error with nodemcu

FrGe2016, 2020-12-28:
Hi

Happy to find someone with potential awnsers to my problem.
I am working with new Chinese css811 sensor with wich i am not able to have connection with the ccs811 library.

I tried unsuccesfully to find a solution in the Web ( using the Adafruit library) 

I  just found and installed your version an hour ago. 
I still have problems CSS811: IC2 error

If i scan the IC2 device i get my css8111 detected at port 0x5A.
My IC2 is working well on my Nodemcu when i plug a BMP280 sensor

What should i do next ?

maarten-pennings, 2020-12-29:
This is very little info you give us.
Could you send me the output of `ccs811basic`?
Could you also show the wiring that you used (schematics or maybe photo if clear enough)?

How did you do the "scan the IC2 device i get my css811 detected at port 0x5A."

FrGe2016, 2020-12-29:
Solved i had a bad solder with intermitant problems.

Thanks for your care

closed 2021-07-10 by maarten-pennings