maarten-pennings/CCS811 #62: LilyGo T5 V2.3 and cjmcu-8128. Unable to flash ccs811

RA-NGR, 2021-01-27:
cjmcu-8128 connected as SDA - 21, SCL - 22, WAKE - 32

Log is:

Starting CCS811 flasher
setup: library     version: 11
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 1100
setup: starting flash of 'CCS811_FW_App_v2-0-0.bin' in 5 seconds

ccs811: ping ok
ccs811: reset ok
ccs811: status (reset1) 10 ok
ccs811: app-erase ok
ccs811: status (app-erase) 10 ERROR
setup: CCS811 flash FAILED

RA-NGR, 2021-01-27:
Solved. Do not use **sciosense/CCS811_driver** - a lot of errors in code.

closed 2021-01-27 by RA-NGR

maarten-pennings, 2021-01-27:
Do you have some examples of errors on the sciosense driver code?

RA-NGR, 2021-01-28:
`bool ScioSense_CCS811::flash(const uint8_t * image, int size) {
  uint8_t sw_reset[]=   {0x11,0xE5,0x72,0x8A};
...
// Invoke a SW reset (bring CCS811 in a know state)
    PRINT(F("ccs811: reset "));
	ok= this->write(_slaveaddr, CCS811_SW_RESET, sw_reset, 5);
...
// Invoke app erase
    PRINT(F("ccs811: app-erase "));
	ok= this->write(_slaveaddr, CCS811_APP_ERASE, app_erase, 5);`
The result is app-erase not executing.
After coorect this flash exits with error and none blocks writed.
I download your code and flash without problems from first attempt