maarten-pennings/CCS811 #63: Sensor fault or soldering issue?

akshulg, 2021-02-02:
I designed a circuit for CCS811 myself using design guidelines given in one of the documents in [sciosense](https://www.sciosense.com/products/environmental-sensors/ccs811-gas-sensor-solution/) website. After assembling when I was testing the sensor for data I was able to receive eco2=400 and etoc=0. By covering the sensor the tvoc values increases and I sprayed liquid hand sanitizer near the sensor area, that also proved the behavioural aspects. When the sensor was ON for 1 hr or so then base value was 40-60ppb. 
`[17:25:38:184] CCS811: eco2=7735 ppm  etvoc=13014 ppb
[17:26:05:241] CCS811: eco2=2164 ppm  etvoc=1058 ppb 
[18:10:30:336] CCS811: eco2=880 ppm  etvoc=73 ppb
[18:16:02:002] CCS811: eco2=728 ppm  etvoc=49 ppb`

The next day I was supposed to compare the CCS811 with my current voc sensor IAQ CORE-C. When I turned CCS811 on I was getting error code. 
`CCS811: errstat=1091=--vHxmrwF--Ad-iE`
After reading datasheet and the .cpp file of the library I understood 3 issues. Heater voltage, heater current and max resistance.
When I was troubleshooting my hardware, I realized that IAQ core and CCS811 have the same I2C address. So i disconnect IAQ core. But even after that CCS811 was giving the error code. I realized that due to I2C address conflict sensor might got damaged. SO I assembled a new PCB and only after 2 data points that new sensor also started giving error code.
`CCS811: eco2=400 ppm  etvoc=0 ppb 
CCS811: eco2=400 ppm  etvoc=0 ppb 
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=899=--vhXmrwF--AD-iE
CCS811: errstat=2899=--VhXmrwF--AD-iE` 

Can this be soldering issue that the sensor keeps giving error codes? What is the best way to solder the sensor?

akshulg, 2021-02-26:
@maarten-pennings any help would be appreciated!

maarten-pennings, 2021-02-26:
I understand your problem, but this is outside my knowledge area. 

I know that many datasheets contain rules for soldering temperature profiles, but I don't know how to do that as an amateur, and I don't see it in the ccs811 datasheets. Did you ever try a ccs811 breakout board, which is already soldered?

But I agree that clashing I2C addresses should not break a sensor.

Are you sure your voltages are always in spec?

closed 2021-07-10 by maarten-pennings