maarten-pennings/CCS811 #64: Issues in CCS811 sensor reading

shaileshbakshi, 2021-02-12:
The following are the issues we are having with the ccs811 sensor.

1) If we are moving a sensor from places where there is a high level of gas to a low level of gas. The sensor is taking a large amount of time to settle down to the actual gas level of that place.

2) We have noticed that sometimes there are sudden jumps in CO2 and TVOC values.

3) If sensors exposed to the high level of Co2 and TVOC  ( Co2: 2000 to 2500 TVOC: 400 to 800) constantly for more than 4 to 5 weeks they are getting damaged after that those sensors are reporting co2 value of nearly around 400 to 500 even if it is exposed to the higher level of gas.

Following is the initialization flow of the Gas sensor

1) Turn on the sensor.
2) Initialize sensor in PulseHeat drive mode (10-second interrupt).  
3) Update temperature and humidity at every 2-second interval.
4) After a reset or a startup Wait for 30 min for the gas sensor to adjust according to the environment and then write the previously-stored baseline value if present else stores the baseline. 

Following Hardware we are using

1) Microcontroller : nRF52832.
2) Temperature and humidity sensor: HTS221.
3) Gas sensor:  CCS811

Please let us know, what is going wrong in the initialization process of the sensor resulting in these issues 

Below is the flow chart for the reference

[image]

maarten-pennings, 2021-02-13:
Hi I'm software developer, and these questions require an expert on MoX sensors.

One thing in your flow chart triggers me, and that is in the areas of baseline management. You correctly wait for the sensor to settle before using reading or writing the baseline. That's good. But it also suggests that you "backup" the baseline from sensor to mcu once, 30 minutes after startup. I believe the application note suggests to backup just after switching off the sensor (if it has been on long enough).

Another question: did you write the humidity and temperature values in the correct format? These are a bit obscure.

shaileshbakshi, 2021-02-15:
Thanks for the reply, I have checked in my Firmware I am writing temperature and humidity in a correct format as mentioned in the datasheet

closed 2021-07-10 by maarten-pennings