maarten-pennings/CCS811 #68: Readings drift higher over time.

redstorm1, 2021-07-20:
setup: Starting CCS811
setup: ccs811 lib  version: 10
setup: hardware    version: 12
setup: bootloader  version: 1000
setup: application version: 2000

Sensors been running on a ESP8266 for over a week, so calibration / baselining should be complete.

The readings start around 400ppm and drive higher overtime too be 1500+ which would be headache inducing.
Im wondering if using firmware 2.01 would solve this or if this is just the issue with these cheap [sensors.]

I am using the wake pin but am thinking of just connecting it to ground to see if the readings stablise to expected values.

```
temp|Hum|CO2|TOVC|DateTime
18.2 | 80.9 | 1708 | 347 | 20/07/2021 3:30 |  
18.2 | 81 | 1687 | 326 | 20/07/2021 3:45 |  
18.1 | 81.1 | 1664 | 303 | 20/07/2021 4:00 |  
18.1 | 81.3 | 1644 | 285 | 20/07/2021 4:15 |  
18.1 | 81.5 | 1632 | 274 | 20/07/2021 4:30 |  
18.1 | 81.4 | 1624 | 266 | 20/07/2021 4:45 |  
18.1 | 81.5 | 1834 | 494 | 20/07/2021 5:00 |  
18.1 | 81.1 | 1790 | 439 | 20/07/2021 5:15 |  
18 | 80.8 | 1775 | 420 | 20/07/2021 5:30 |  
18 | 81 | 1754 | 395 | 20/07/2021 5:45 |  
18 | 80.8 | 1738 | 374 | 20/07/2021 6:00 |  
17.9 | 80.6 | 1748 | 389 | 20/07/2021 6:15 |  
17.9 | 80.5 | 1728 | 365 | 20/07/2021 6:30 |  
17.9 | 80.3 | 1704 | 342 | 20/07/2021 6:45 |  
17.8 | 80 | 1721 | 359 | 20/07/2021 7:00 |  
17.8 | 79.7 | 1687 | 326 | 20/07/2021 7:15 |  
17.8 | 79.7 | 1761 | 404 | 20/07/2021 7:30 |  
17.7 | 79.7 | 1824 | 480 | 20/07/2021 7:45 |  
17.8 | 79.8 | 1834 | 494 | 20/07/2021 8:00 |  
17.8 | 80 | 1801 | 451 | 20/07/2021 8:15 |  
18.5 | 80.6 | 1720 | 357 | 20/07/2021 8:30 |  
19.1 | 78.6 | 1651 | 291 | 20/07/2021 8:45 |  
19.2 | 78.3 | 1591 | 236 | 20/07/2021 9:00 |  
21.1 | 76.9 | 455 | 8 | 20/07/2021 9:15 | Rebooted
22.6 | 74.7 | 486 | 13 | 20/07/2021 9:30 |  
23.7 | 72.9 | 418 | 2 | 20/07/2021 9:45 |  
24.5 | 71.9 | 448 | 7 | 20/07/2021 10:00 |  
25.4 | 70.5 | 423 | 3 | 20/07/2021 10:15 |  
26.3 | 68.7 | 418 | 2 | 20/07/2021 10:30 |  
25.3 | 69.7 | 456 | 8 | 20/07/2021 10:45 |  
24.3 | 68.5 | 561 | 24 | 20/07/2021 11:00 |  
24.1 | 70.3 | 540 | 21 | 20/07/2021 11:15 |  
23.9 | 70.8 | 551 | 23 | 20/07/2021 11:30 |  
23.5 | 72.4 | 578 | 27 | 20/07/2021 11:45 |  
23.5 | 72.4 | 553 | 23 | 20/07/2021 12:00 |  
```

Osfuco, 2021-08-20:
Did you found any solution?

iCassius, 2021-09-27:
same issus
and i was looking for that how to get and set baseline at init

maarten-pennings, 2021-09-27:
There is an app note about that:  https://www.sciosense.com/wp-content/uploads/documents/Application-Note-Baseline-Save-and-Restore-on-CCS811.pdf