maarten-pennings/CCS811 #69: CCS811 Baseline Issues

kdonahoe, 2021-10-08:
Hello,

I've been having baseline issues with my CCS811 sensor for a while. Below is a plot where the blue line is VOC, and red is the baseline value, scaled down (/100) to fit on the plot. During the day, there is a lot of activity, which is expected, but after 5 or 6 PM, no people or chemicals should really be in the area. The VOC value drifts pretty crazily during the night hours, and this has been witnessed for weeks (it is running continuosly). Currently writing the temp and humidity to the ENV registers, but not writing to the baseline at all, because that behaved oddly for me in the past. My questions are:

1) Why do the baseline values change so much during stable (almost zero) VOC values?
2) How do we turn off automatic baseline correction?
3) Recommended fix?

[image]

Thanks for any help

iCassius, 2021-11-04:
I have same issus . I bought 2 ccs811 with 2 nodemcu for testing.
One uses esphome and the other uses library
One of the readings will start to rise indefinitely after a random time
The value of TVOC and Co2 will rise to more than 30000+
[image]
I got the Baseline and re-written it, the value will return to 400+ after restarting

maarten-pennings, 2021-11-05:
I expect that the software just reads the values measured by the css811 chip. So I expect that the differences are caused by the chips. That you try swapping them to see what the differences come from?

nordicp, 2021-11-08:
TVOC and eCO2 output is processed data and without looking at the sensor resistance it is hard to make a guess. Please log the two sensor aside each other and record eCO2/TVOC and sensor resistance.
Do both modules have the same history (total power up time, same mode of operation) ?
I recommend mode 1 (1s data interval) for best performance.

JeeWee65, 2023-02-02:
> TVOC and eCO2 output is processed data and without looking at the sensor resistance it is hard to make a guess. Please log the two sensor aside each other and record eCO2/TVOC and sensor resistance. Data graphs of resistance should have a logarithmic y-axis.
> Do both modules have the same history (total power up time, same mode of operation) ?
> I recommend mode 1 (1s data interval) for best performance.

I have same issues as iCassius. How can I get the resistance in a graph? Might be obvious to you but I didn't see code how to read that yet.

maarten-pennings, 2023-02-02:
The last output parameter of the read() method returns the raw data. One part of that is the electrical current. The other part is an adc value which maps to a voltage. From current and voltage you can compute resistance. Hope this helps enough.

JeeWee65, 2023-02-03:
> The last output parameter of the read() method returns the raw data. One part of that is the electrical current. The other part is an adc value which maps to a voltage. From current and voltage you can compute resistance. Hope this helps enough.

Thank you. I have the sensor hooked up via a D1 mini to ESPHome to get and use the data in Home Assistant. In ESPhome I don't sees this raw data, I see however that the baseline shifts with the CO2 value although I've set a baseline value in the configuration.

If I don't find another solution I will try to connect it to my pc and try the Arduino code.