maarten-pennings/CCS811 #71: Can it really be used for co2 monitoring?

raomin, 2021-12-15:
Hi @maarten-pennings, 
Thanks for your work and dedication on this.

I'll be working on project to equip class rooms with _air quality_ sensors. To test the accuracy of the CCS811, I made it run next to a MH-Z19b that is more expensive but proven to be accurate for CO2 values. The CCS811 module has an HDC1080 and my code updates its environmental data.

After several days running non stop in my room (`CCS811_MODE_1SEC`), the compared graph of the co2 values for (eco2) CCS811 vs MH-Z19b shows a completely erratic result for CCS811.

[image]

My questions are:

- I know CCS811 only provides an estimated CO2 but here it's far off. Do you think CCS811 can be more accurate for CO2?
- If not, are there studies/tests that validates its accuracy/usefulness for general air quality (etvoc)? (in other terms, shall I keep in for this project....)

Thanks!

maarten-pennings, 2021-12-16:
Hi, what can I say...
The CCS811 has NO capabilities to measure CO2. Period. The CCS811 is a MoX sensor and it does NOT react with CO2.
The CCS811 does react with VOCs (Volatile organic compounds), stuff people produce (eg breath out).
The idea is that the CCS811 is used _indoor_ so that the variation in VOCs has human origin only and can be used as a guidance to estimate CO2. Your graph is not even that bad - it follows the green line quite well. It would have been better if the red line was divided by 2.
This is an engineering decision: small, low-cost, low power (CCS811) versus accurate (real CO2 sensors).

raomin, 2021-12-18:
Thanks for this clarification, it all makes sense. I'll measure the CO2 with a CO2 sensor and keep the ccs811 for VOC.

closed 2021-12-18 by raomin