maarten-pennings/CCS811 #72: C811 baseline

kwoerdehoff, 2022-10-03:
Hello!
Made a Projekt with the CJMCU-811 (C811 with HDC1080, Temp&Humid).
Made a Sketch with your Library and its running so far.
While running the Datas about eCo2 and eTVOC is getting higher and higher and jumps up and down.
So i tried to get Informations about this. I found an german Article about the C811 and its "Baseline".
Is there any CodeExample for putting and rerwiting the Baseline in and out of the EPROM?
Iám no Programmer, so i need an Example to put it in my Code.
Would be verry nice to get Help.

;_)) Klaus

maarten-pennings, 2022-10-04:
Hi,

To save and restore the baseline is easy.
The hard part is _when_.
There are several issues about baseline management in this GitHub project.
For example, see https://github.com/maarten-pennings/CCS811/issues/22#issuecomment-546696611

kwoerdehoff, 2022-10-14:
Hello Maarten,

have been on Vacation.

Thanks a lot for your help.

Will look at it and tell you what happens.

 

Best Regards ;_)) Klaus

 
 

Gesendet: Dienstag, 04. Oktober 2022 um 22:31 Uhr

 

For example, see #22 (comment)